FreeCAD/FreeCAD #14003: [TreeView] Incorrect Label2 view when element is hiden

Kuzma30, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When user hide Object (by pressing space) in TreeView  text in Label2 column shown as normal text.
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Unknown
Branch: (HEAD відʼєднано на 0e24e121eb)
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * fasteners 0.5.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
What is the expected behavior? Without the visibility icon active, show it greyed out as the object label?

Kuzma30, 2024-05-14:
I think when visibility icon non active show all columns greyed as the object label.

Kuzma30, 2024-05-15:
```
diff --git a/src/Gui/Tree.cpp b/src/Gui/Tree.cpp
index 958aacf170..29c8f3d85c 100644
--- a/src/Gui/Tree.cpp
+++ b/src/Gui/Tree.cpp
@@ -2690,20 +2690,20 @@ void TreeWidget::sortDroppedObjects(TargetItemInfo& targetInfo, std::vector<App:
 
 void TreeWidget::drawRow(QPainter* painter, const QStyleOptionViewItem& options, const QModelIndex& index) const
 {
-    QTreeWidget::drawRow(painter, options, index);
+    //QTreeWidget::drawRow(painter, options, index);^M
     // Set the text and highlighted text color of a hidden object to a dark
-    //QTreeWidgetItem * item = itemFromIndex(index);
-    //if (item->type() == ObjectType && !(static_cast<DocumentObjectItem*>(item)->previousStatus & 1)) {
-    //    QStyleOptionViewItem opt(options);
-    //    opt.state ^= QStyle::State_Enabled;
-    //    QColor c = opt.palette.color(QPalette::Inactive, QPalette::Dark);
-    //    opt.palette.setColor(QPalette::Inactive, QPalette::Text, c);
-    //    opt.palette.setColor(QPalette::Inactive, QPalette::HighlightedText, c);
-    //    QTreeWidget::drawRow(painter, opt, index);
-    //}
-    //else {
-    //    QTreeWidget::drawRow(painter, options, index);
-    //}
+    QTreeWidgetItem * item = itemFromIndex(index);^M
+    if (!isVisibilityIconEnabled() && item->type() == ObjectType && !(static_cast<DocumentObjectItem*>(item)->previousStatus & 1)) {^M
+       QStyleOptionViewItem opt(options);^M
+       opt.state ^= QStyle::State_Enabled;^M
+       QColor c = opt.palette.color(QPalette::Inactive, QPalette::Dark);^M
+       opt.palette.setColor(QPalette::Inactive, QPalette::Text, c);^M
+       opt.palette.setColor(QPalette::Inactive, QPalette::HighlightedText, c);^M
+       QTreeWidget::drawRow(painter, opt, index);^M
+    }^M
+    else {^M
+       QTreeWidget::drawRow(painter, options, index);^M
+    }^M
 }

```
I found this commented code and make small modification to it (add !isVisibilityIconEnabled()  )
[image]
[image]

PR #14075 (2024-05-20): Gray out all columns in the tree if an item is invisible
Fixes #14003.
closed 2024-05-20 by chennes