FreeCAD/FreeCAD #14005: Sketcher: "Fit All" zoom command always includes axis origin

furgo16, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Open the attached file
2. Double-click on the `Sketch` sketch to edit it
3. Click on the [Std_ViewFitAll](https://wiki.freecad.org/Std_ViewFitAll) toolbar button, or use its shortcut <kbd>V</kbd>, <kbd>F</kbd>
4. **Expected**: the view is resized, and the rectangle occupies most of the view
5. **Actual**: the view is resized, yet the rectangle occupies only a small corner of the view. The command seems to always want to include the origin axis, where there seems to be a construction point.

[image]
_Before Fit All_

[image]
_After Fit All_

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git) Snap 1019
Build type: Release
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
If you dimension the object (fully constrained) you somehow need to include the axis or any reference, right?

furgo16, 2024-05-14:
Good point. I'm far from an expert, but from my point of view:
- This particular sketch is not fully constrained, so there is no need to include any axis in the zoomed view
- Even if it would be constrained to an axis, why not show the relevant axis then? At this time it seemingly zooms to include the 0, 0 axis point, which seems to have a construction point on it.

furgo16, 2024-05-14:
Something to note is that this is not something new with FreeCAD 0.22. As far as I can tell, 0.21 behaves equally.