FreeCAD/FreeCAD #14007: Sketcher: "Fit selection" command in edit mode does not fit view to selection

furgo16, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Open the attached file
2. Double-click on the `Sketch` sketch to edit it
3. Select the four lines of the rectangle
4. Click on the [Std_ViewFitSelectionl](https://wiki.freecad.org/Std_ViewFitSelection) toolbar button, or use its shortcut <kbd>V</kbd>, <kbd>S</kbd>
5. **Expected**: the view is resized, and the rectangle occupies most of the view
6. **Actual**: the view is not resized, nothing happens.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git) Snap 1019
Build type: Release
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
I guess fit selection is looking at the selected objects in the tree view, not the elements in the sketch, @PaddleStroke can you confirm?

0penBrain, 2024-05-14:
"Fit selection" is not working correctly even selecting an edge of sketch (not in edit mode). :wink:

PaddleStroke, 2024-05-15:
I can confirm that the fit selection does not take into account the subnames (edge, point, face). It is also true with a solid, it will not fit selected edge, only the whole object.

PaddleStroke, 2026-03-22:
Still relevant