FreeCAD/FreeCAD #14016: Error on create output for helical cut on fanuc code

RodrigoTMorais, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When use helical are used in a job, error ocur when post processor is clicked. no code was generated.

File "C:\Program Files\FreeCAD_weekly-builds-37083-conda-Windows-x86_64-py311\Mod\CAM\Path\Post\Processor.py", line 80, in export
    return self.script.export(obj, filename, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD_weekly-builds-37083-conda-Windows-x86_64-py311\Mod/CAM/Path/Post/scripts\fanuc_post.py", line 278, in export
    gcode += parse(obj)
             ^^^^^^^^^^
  File "C:\Program Files\FreeCAD_weekly-builds-37083-conda-Windows-x86_64-py311\Mod/CAM/Path/Post/scripts\fanuc_post.py", line 436, in parse
    if pathobj.Tool.ToolType == "Tap":
       ^^^^^^^^^^^^^^^^^^^^^

'FeaturePython' object has no attribute 'ToolType'

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37083 (Git)
Build type: Release
Branch: main
Hash: 502f78e111c761cb92c51222605b26b8701fcbca
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)

This erro ocurs in FREECAD 0.21.2, 0.20 too
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
Please provide a demo file or steps to reproduce this

RodrigoTMorais, 2024-05-14:
in part design wb, dar an part with a hole.
start job with CAM WB
select fanuc processor
use helical tool to mill the hole
you can simulate in CAM and works fine
click post process, then error will appear.

note: with other processors the error does not occur.

Attached I send the file where the problem appears.

maxwxyz, 2024-05-16:
I cannot reproduce this with your description. Can you upload a .zip file of the FCStd file directly to GitHub?

luzpaz, 2024-07-12:
@RodrigoTMorais please reply in [github](https://github.com/FreeCAD/FreeCAD/issues/14016) (not via email). Compress the FCStd file and drag drop it into the textbox.

RodrigoTMorais, 2024-07-12:
Ok, I will do that.

maxwxyz, 2024-07-14:
@RodrigoTMorais any news on this?

RodrigoTMorais, 2024-07-14:
Returns error when clicking to generate gcode when FANUC is selected, with other formats it works fine. tested on FreeCad 0.21.2 and FreeCad 0.22.dev (rev37645).

[cam03.zip](https://github.com/user-attachments/files/16226685/cam03.zip)

darrencocco, 2025-02-17:
Hey this is still a bug in 1.0 FYI but I've found out how to actually get around the problem.

Please see in the attached image the highlighted Property I've manually added 3D model for the toolbit.
It's called ToolType and it satisfies the requirements in the fanuc post processor. 

[image]

I've noticed that other parts of the code also make a reference to ToolType but then fall back to ShapeName if the attribute/property isn't present.
There seems to be a UI definition for "toolType" in the ToolEditor.ui but I couldn't for the life of me find the corresponding QComboBox in the FreeCAD ui.
See here for all the tooltype references: https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20tooltype&type=code

As for how to fix this I don't really have the an idea as to what the right solution is.
* Do we fall back to ShapeName like the other parts of the CAM code?
* Do we by default point ToolType to ShapeName and then allow the?
* Do we document it in the Fanuc post processor documentation that ToolType needs to be defined with a special callout to behaviour when tapping?
* Something else?

tarman3, 2025-09-14:
Probably no
At least I can not reproduce the issue with FCStd file attached above

petterreinholdtsen, 2025-11-30:
This issue only trigger when the generated G code include a M3 instruction.  It is the same issue as #25723, and a fix for it is provided in #25799.  I believe this is a blocker for the 1.1 release, as almost any generated G code include starting the spindle rotation.