FreeCAD/FreeCAD #14020: Sketcher: ConstraintEqual doesn't accept Line + Arc

FlachyJoe, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We can't make a Line equal an Arc length without calculation.

### Full version info

```shell
All version
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-15:
What should the equal constrain do in this case? 
Arc radius equal to line length? Diameter to line length? Arc length to line legth? I think this behavior would not be clear to the user. Same with the arcs of hyperbola/parabola.

FlachyJoe, 2024-05-15:
@maxwxyz radius and diameter are available throw construction lines. Only arc length to line length is needed.

pierreporte, 2024-08-31:
This is doable with an expression.