FreeCAD/FreeCAD #14021: Core: "Go to selection" not working

maxwxyz, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Std_TreeSelection ("Go to selection") is intended to scroll to the selected object in the tree view.
This does not work if the object is a nested object, e.g. a body or solid in a part container, or nested arch objects.
If you have a file with nested objects the command seems to do nothing. You have to manually expand/open all arent objects that the behavior works as intended.

The expected behavior is:

- expand all parent objects of the selected object
- scroll to the selected object in the tree view

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * fasteners 0.5.20
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

farleyrunkel, 2024-05-15:
Hello, @maxwxyz, I can reproduct it.
And I can try to fix this problem with expected behavior.

Roy-043, 2024-05-15:
The current behavior depends on Std_TreeSyncSelection. I am not sure if it is intentional.
https://wiki.freecad.org/Std_TreeSelection

closed 2024-08-25 by maxwxyz

semhustej, 2025-04-14:
I just wanted to report the same issue and I found this one. I would vote for changing the behavior to what OP is proposing:

After `Std_TreeSelection` is invoked:
-  expand all parent objects of the selected object
-  scroll to the selected object in the tree view

-----------------------
**Reason for change**
As a user, if I select that I want to select object in the Project Tree, it means that I don't want to go through project tree a search for it manually. I am willing to take the 'risk' that some groups in project tree might be expanded.
I am not a long time FreeCAD user, so I guess there are reasons why the tool does not expand the tree, but I can not think of any use case where this would be desriable.

In my opinion `Std_TreeSyncSelection` is not a replacement of `Std_TreeSelection` in this case, because it is not easy to turn it on only for one object at a time. And when `Std_TreeSyncSelection` is turned on all the time, it basically means that all groups/parents in the project tree are expanded all the time, as user selects objects in 3D view. This makes the project tree very puzzling.

These are the reasons I think the behavior of `Std_TreeSelection` should be change as is proposed in OP.

--------------------
**Alternative solution**
If changing the behavior is not viable, I think there should be at least a yellow warning in the `Report View` in case the tool does not select the object in project tree because it's parent is not expanded. This would inform the user what has happened. Without this warning it is very confusing for new users - the first thought is that the tool is not working.
It is not a good user experience to run a tool without the desired result and no visible clue of why nothing happened.

maxwxyz, 2025-05-31:
Indeed, if the user actively clicks on the command "Go to selection" it should always expand and go to the selection, as otherwise nothing happens. And with the `always expand` setting is is always messing around the tree.
@farleyrunkel are you still interested in creating a PR for this?

ErtugrulEge, 2025-08-02:
I'm so glad it was added to 1.1 milestone. I record a video how painful if you need to find object folder.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

https://github.com/user-attachments/assets/07a1d034-44fe-4fde-97d6-21c0dae11313

maxwxyz, 2025-09-18:
@farleyrunkel feel free to pick it up anytime

cmoliverio, 2025-09-19:
Can I take a stab at this?  I also want this feature. 

Can a FreeCAD developer reach out to me about optimizing workflow? I don't want to recompile FreeCAD for 35 minutes to test a small change in code.  Direction would be appreciated.