FreeCAD/FreeCAD #14027: Hatch not correctly exported to PDF

mpfusion, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The PDF export lacks the 2nd hatch in the following example:

TechDraw window:
[image]

SVG export:
[image]

PDF export:
[image]

To reproduce:

- open attached file
  [missinghatch.tar.gz](https://github.com/FreeCAD/FreeCAD/files/15319511/missinghatch.tar.gz)
- open TD
- Export PDF or Print All Pages

Forum: https://forum.freecad.org/viewtopic.php?t=87667

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Release
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * BOLTSFC
  * kicadStepUpMod 11.1.5
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mpfusion, 2024-05-15:
Seems related, indeed. However, I couldn't find an open bug report for that issue.

maxwxyz, 2024-05-15:
could this be related to an older issue (closed) https://github.com/FreeCAD/FreeCAD/issues/11088

WandererFan, 2024-05-19:
related issue: TechDraw: Hatching clip path functionality is missing from svg export #13338

mpfusion, 2024-05-19:
As I noted in the OP the SVG export is fine, it's the _PDF_ export that's having an issue. But that issue might still be related, though.

closed 2024-10-19 by WandererFan