FreeCAD/FreeCAD #14036: PartDesign: Most subtractive features use the same task panel icons and descriptions as their additive counterparts

FEA-eng, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the subtractive PD features other than Pocket use the same icons and descriptions in their task panels as their additive versions. Groove, for example:

[image]

It should be changed for consistency. Unfortunately, as pointed out in [this comment](https://github.com/FreeCAD/FreeCAD/pull/13968#issuecomment-2111479551), it may not be so easy.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-15:
@marcuspollio FYI

marcuspollio, 2024-05-15:
> @marcuspollio FYI

Thanks for creating this issue.
Part Design Task Panels UI code would indeed deserve some refactoring so their content could be more appropriate to each and every feature.
That's sadly out of my reach (know nothing about C++)
If someone steps in for that, I can help for content and testing.

luzpaz, 2024-08-11:
@FreeCAD/design-working-group can someone weigh-in regarding ideas or persons that would be interested in working on this ?

pierreporte, 2024-08-11:
It would make sense to have the actual feature icon. I thought it was done recently.