FreeCAD/FreeCAD #14037: Wrong styling for task panel fold button

Roy-043, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The buttons that allow (un)folding task panel sections are wrongly styled in the most recent dev version (Classic theme on Windows).
[image]

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-05-15:
I can confirm it. Related to #13697 ?

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

FEA-eng, 2024-05-27:
@MisterMakerNL Can you take a look at this? It definitely needs to be fixed for 1.0. Maybe even #13697 should be reverted?

[image]

MisterMakerNL, 2024-05-27:
Why oh why does this not happen when you use a stylesheet?
Anyway, is easy fix since I can export these images from an svg file.

closed 2024-06-03 by chennes