FreeCAD/FreeCAD #14038: TechDraw: Area dimension doesn't work the first time it's applied

FEA-eng, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

#13525 introduced the great new smart dimension tool but also the individual area dimension tool. The former works perfectly but the latter fails the first time it's applied. Steps to reproduce:
1. Add Part Cube.
2. Add TechDraw view of the cube.
3. Select the face and apply the Area Dimension (not the old Area Annotation).

It shows 0 mm^2 and the following error message:

`<Exception> getGeomTypeFromName - empty geometry name`

Just retrying should make it work then.

Btw. do we still need the area annotation? If yes, these 2 tools should have different icons:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-15:
@PaddleStroke FYI

FEA-eng, 2024-06-29:
@maxwxyz Can you add the 1.0 label? The tool doesn't work properly.

PaddleStroke, 2024-07-01:
I have tested and this is not specific to area dimension. It's a bug for all type of dimension.
What happens is that when you selected the face, the object itself is still selected. So when you add the dimension the part ends up in 'reference 3d' : 
[image]

Removing it and the dimension works.

PaddleStroke, 2024-07-01:
@WandererFan can you please weight in what needs to be done here?
https://github.com/FreeCAD/FreeCAD/assets/19984177/50f9c2fd-7370-406b-98b3-2f349f6e2ade

PaddleStroke, 2024-07-01:
Ok so I checked it anyway. And so getReferencesFromSelection is putting the solid object (part in my case) into references3d. Which is not cleared and is added by setReferences3d.

PaddleStroke, 2024-07-01:
Ok so the fix seems pretty simple. Just add 

```
    else {
        references3d.clear();
    }
```

to the if : 
```
    if (geometryRefs2d == TechDraw::isViewReference && !references3d.empty()) {
        ...
    }
    else {
        references3d.clear();
    }
```

@WandererFan please check if this solution is OK.

FEA-eng, 2024-07-01:
@PaddleStroke Thanks for taking care of this. What do you think about the second part of the issue - duplicated icons/tools?

PaddleStroke, 2024-07-01:
Removing stuff has proved to be tedious in the past so I prefered to just skip it.
But I agree it makes not much sense keeping the annotation tool.

FEA-eng, 2024-07-01:
> Removing stuff has proved to be tedious in the past so I prefered to just skip it. But I agree it makes not much sense keeping the annotation tool.

Maybe for now, they should at least have slightly different icons. It's confusing when they are identical.

Reqrefusion, 2024-07-03:
#14878 Since it is mentioned here, let me write that with the icon revision, a difference in nuance was added between the icons. But this difference is not enough to overcome this confusion.

PR #15084 (2024-07-08): TechDraw: Fix 14038 and Refactor CommandCreateDims 
- Fix #14038 by adding 
```
    else {
        references3d.clear();
    }
```
to make sure no unwanted reference3d is added.
- Refactor CommandCreateDims by adding execDim to reduce code duplicity.