FreeCAD/FreeCAD #14039: Start page template buttons may require 2 clicks

FEA-eng, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's not a big issue and it happens only once in a few times but can be quite annoying. I've noticed that the Start page template buttons may not work the first time you click on them. They just get the green outline but nothing else happens. Only clicking the second time activates the template. I've noticed it with the Empty file template because I use it very often but I've seen it at least with the 2D Draft template as well.

It seems that it happens when the buttons are clicked too quickly after starting FreeCAD.

https://github.com/FreeCAD/FreeCAD/assets/59876896/b0996a1e-aaef-41f6-bca5-6395c5c8e834

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-15:
Is different to #13991?

FEA-eng, 2024-05-15:
> Is different to #13991?

It might be the same bug but I'm not sure. Clicking on the button's title doesn't work for me but clicking on the description and anywhere else does. In the video, I didn't click on the title (or even description) itself but maybe close enough.

maxwxyz, 2024-05-16:
@chennes FYI

Syres916, 2024-05-17:
@FEA-eng I'm using a stylesheet (Behave-dark) and found this to be really annoying so I had a look in the code and found that if I added the boolean parameter _FileCardUseStyleSheet_ to _BaseApp/Preferences/Mod/Start_ and set to _false_ that my QPushbutton settings from Behave-darkwould be wholly put in use. This then allows the entire button to be clicked without any issue. Now it looks like you're using Classic without stylesheet so it may not work for you but worth trying.

Now there's something that should prevent all this and that's https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/Gui/StartView.cpp#L227 but on my Debian 12.4 system this doesn't seem to pick up any stylesheet neither a FC provided or system one from Cinnamon. In some previous work on Techdraw I ended up using my own function to correctly pick my stylesheet up.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * Design456 0.0.1
  * Rocket 3.3.0
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * fasteners 0.5.20
  * freecad.gears 1.1.0
```

FEA-eng, 2024-05-18:
> Now it looks like you're using Classic without stylesheet so it may not work for you but worth trying.

Yeah, unfortunately, this doesn't work for me.

Syres916, 2024-05-21:
> Yeah, unfortunately, this doesn't work for me.

OK, I've been tracking down the root cause and it appears this is all down to the Html bold tags so I've removed the heading and made the description bold using the QLabel style. You will have to compile https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_New_User_Start_Buttons and retest. It works for me as a completely new user.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37431 (Git)
Build type: Release
Branch: main
Hash: 1eac717f8f6d9471f88f6af3a3f189cab6fba84c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * Design456 0.0.1
  * Rocket 3.3.0
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * fasteners 0.5.20
  * freecad.gears 1.1.0

```

Syres916, 2024-05-24:
@FEA-eng are you not able to compile and test the above branch, it would be quite good to get this improved?

FEA-eng, 2024-05-24:
> @FEA-eng are you not able to compile and test the above branch, it would be quite good to get this improved?

I can compile it, I just didn't have time to do it.

FEA-eng, 2024-06-29:
@maxwxyz Can you mark this one for 1.0 too? It's IMO the worst Start page issue and really annoying in general.

commit db3358d5 (2024-08-12): Make start page use QStackedWidget + other fixes

closes #14039, closes #14694, closes #14817, closes #15113
closed 2024-08-12 by chennes