FreeCAD/FreeCAD #14040: Sketcher B-splines hate symmetry at endpoints

krushia, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher solver will oscillate between "Over constrained" and "Under constrained" when B-splines have a symmetry and a coincident constraint at the end. Easiest to reproduce with ends on arcs with additional tangent constraints (not convinced a tangent constraint is required to trigger it though).

The attached test case shows as okay in this screenshot, but if you try to drag b-splines around it'll claim redundancy about every other twiddle.
[image]
[bspline-solver-buggies.zip](https://github.com/FreeCAD/FreeCAD/files/15326888/bspline-solver-buggies.zip)

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Sketcher

### Anything else?

Possibly related are https://github.com/FreeCAD/FreeCAD/issues/10248 and https://github.com/FreeCAD/FreeCAD/issues/12653 and https://github.com/FreeCAD/FreeCAD/issues/13645 and https://github.com/FreeCAD/FreeCAD/issues/13930

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-15:
Tried to use the "Sketcher_JoinCurves" tool to make the same geometry, and ended up with the error `<class 'ValueError'>: Connecting the end points of the same curve is not yet supported.`

I wonder if that sums up the core issue, which would what #12653 is about. In that case, this bug should maybe be titled "Sketcher doesn't warn about unsupported loops of curves in all cases"?

maxwxyz, 2024-05-16:
@AjinkyaDahale this seems exactly like #12653 can you confirm?

AjinkyaDahale, 2024-05-17:
Probably a generalization: in #12653 the issue is related to a B-spline being end to end tangent with itself.