FreeCAD/FreeCAD #14042: Feature proposal: user defined (Python?) functions for Expressions and Spreadsheets

wolfgangr, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From elaborated office spreadsheets (and other applications), we know that we can supply **user defined functions** to **expand the functionality** of the formula language in customized manner.

In the FreeCAD envrionment, I'd assume that **Python** were the language of choice for such an endeavour.

In @realthunder 's Wiki
https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet
there even is described an elaborated ability to acces **python in Expressions** (and thus in spreadsheets, too).

However, I **failed to reproduce** this on current stable 0.21.2.
- Was this feature merged already?
- Did I just miss the proper documentation how to use it?
- Is there some hidden way to switch this on?
- Is this feature available in current dev? 
- Are there any plans in the future to add it?
- Is there any other way to extend the expression engine at user level?

On my journey from bloody novice to still-bloody silly-user in FreeCAD, there were two occasions where I really missed user defined functions:

- "**Law of cosine**" to calculate the **angles in a triangle**, provided three lengths, to nail down flipping sketches
https://forum.freecad.org/viewtopic.php?t=86102
- Access to global placement of **linked items,** which is **broken** in the "**globalPlacement**" as accessible by Expressions/Spreadsheets, but available by workaround in Python
https://forum.freecad.org/viewtopic.php?p=759225#p759225

Of course, this are mere examples, other people may have other needs in mind.

From my experience, user defined functions (e.g. in Excel) reduce the amount of programming and object integration required to a considerable extent, compared to implement a far wider superset of the functionality in high level language, as it seems the way to go with FreeCAD at the moment - see e.g.
https://wiki.freecad.org/Python_scripting_tutorial

### Full version info

```shell
[code]
OS: Debian GNU/Linux 10 (buster) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.17
  * slic3r-tools
  * Pyramids-and-Polyhedrons
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.12
  * FeedsAndSpeeds 0.5.0
  * dodo 1.0.1
  * frame 0.1.1
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * MOOC 2022.4.21
  * Help 1.0.3
  * ose-piping
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.11
  * ThreadProfile 1.89.0
  * SteelColumn
  * Estimate 0.1.2
  * DynamicData 2.60.0
  * parts_library
  * btl 0.9.9
  * AnimationFreeCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.8
[/code]
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-16:
- Feature hasn't been merged yet. TNP fixes are happening first.
- You didn't miss any documentation.
- No hidden switch.
- Not in dev either.
- Go here: https://github.com/FreeCAD/FreeCAD/pull/3062
- Expressions can't be extended without coding and recompiling, no. Expressions aren't at the Python level.

It only works in realthunder's Link branch for now unfortunately.

I've taken notice of your forum threads while on my own FreeCAD learning journey. We seem to have similar frustrations regarding limitations of expressions and general disconnect between the GUI and the object-oriented base it rests on.

IMHO FreeCAD really needs a *safe* scripting system (like Lua?) that allows us to do some level of programming that gets saved in the FCStd file. That or a way to put Python macros in it and ask the user if its okay to run them. Anyway, you are not alone.

Zolko-123, 2024-05-16:
Embedding Python code into a FCStd file has been discussed many times and is refused for security reasons : to separate data and code. If you want to ship your own code there are already many available methods (macro, FC object ...)

pitch-circle, 2024-05-16:
> there were two occasions where I really missed user defined functions:

Having two occasions where something different would help you is quite a long way from justifying opening up Expressions to full-on Python. To me, the proposed solution is overkill for the problems. Also, the two issues you identify are quite different in scope.

Cosine rule can already be done with Expressions, it just needs a little more typing.

The global placement issue is different and might be fixed locally by opening up just that part which is currently closed to Expressions.

What I am saying is that until the RT changes reach the top of the in tray, small wins in focussed or targeted places might be easier to achieve.

On cosine rule, please see https://github.com/sbyrnes321/trianglesolver A similar solve() function might be easier to add to the list of permitted Expressions (cube root was added recently as a standalone function, even though exponentiation to 1/3 has long been possible).

wolfgangr, 2024-05-16:
It's **not** about **begging for a fish** (or two), but about **empowering users to do fishing** on their own, in general.

Regarding **choice of language**, I think, **any Turing complete language** might do. May be even some superset of the current expression language, complemented by some control structure and skd of "define function"?

My favour for **Python** was due to the **native access to object properties** - which leads to the question of **security**.
Might it ease the concerns to **limit access to read-only**, and may be only to objects visible in the tree of the **current file**?
I have no clue about the architectural nuts&bolts of Python. Whether it were possible to limit part of it in a "**sandbox**" like way. Maybe it would require an independent instance of python (which we would need for e.g Lua anyway).

I understand that the development carawane has priority towards TNP.
It's FOSS, anyway, and in the end, every feature has to be honoured as a gift, of course.

But coming new from outside, I see the FreeCAD communty falling into two parties regarding TNP:

- avoid it
- attack it

Why do associations of primordial reptile instincts come to my mind? Humans have learned to overcome such instincts with rational decisions. But they are still humans, prone to be caught by emotional patterns, I know....

E.g. there is a sketching wiki (I consider flipping sketches the little sister of TNP, OK?), suggesting to avoid datum constraints as fas as possible. And there is a TNP wiki, suggesting the opposite: relying on datum constraints as far as possible, to keep control on the model.

From a sportive or hunting spirit, I'm with the TNP attackers: Failure is not an option. We'll do it, because we can do it. And FreeCAD is going to be THE one FOSS CAD, so there is no room for compromise. OK.

I also see that toddler like attachment aka "wanna this, there, that way..." appears fast, efficient and attractive, in the first moment. But many time I have allowed myself to be lured down that way, I had to pay long hours of repairing my designs, as soon as I wantet to apply some minor modifications.

There are people who just want to do some mediocre designs. Like me, but not only. Have their sportive goals in other areas. Instead of elaborate ways to cope with inevitable ambiguities resulting from the behaviour of sophisticated solvers, they tend to avoid those solvers all togehter, as long as they don't se a real advantage for their specific design.

People like me, who start to keep track of all placement of assembled subelements in a spreadsheet now, since, as soon as I loose control, I have hard times to gain it back. Cumbersome, but safe. And in the end much faster than reparing a complex model that has fallen apart due to uncomprehensible opaque automagics.

I understand that at the moment, there is no way to implement my idea. Can live with that.
But why not just keep this issue as a collector of ideas and whishes?

So, I'd suggest that some authorized developer might rate this issue as "delayed but not forgotten forever"?
Thank You :-)

adrianinsaval, 2024-05-27:
python in expressions would be a security nightmare so I don't think we'll do that. User defined functions could be a desirable feature

wolfgangr, 2024-06-02:
Have drafted a `Spreadsheet::SheetPython` with user definable functions:
https://github.com/wolfgangr/myTinyAsm/blob/master/sheetPythonCustomFunc.py
Not yet perfect, no documentation, but basically working.

Even used this as a test case:
> On cosine rule, please see https://github.com/sbyrnes321/trianglesolver

Could bind the `trianglesolver` into my spreadsheet, referring to `Sketch.Constraints`, without any extra Python code. All Parameter wrapping purely in FreeCAD Expressions.

> ... security nightmare ...

I understand and accept the decision of the main developer team not to open that door explicitly in core.
Nevertheless the **door is alredy open**:
As my example prooved, as soon as a user decides to import a Macro or whatever Python code, there is always the possibility that some **'evil `eval`'** is in there.
So I'd rather to try to **keep it as save as possible**.

- Can we skd of '**sandbox**' the execution envionment of an **`eval`** call?
- there are optional `global` and `local` dictionaries - my that help?
- is a restricted namespace really a security barrier at all? Or 'only' some help aganist trapping into unchartered pitholes?
- how might a clear and easy interface look like, so that even a Python novice keeps control?

I consider two configuration fields (implemented as Properties):
- a list of directories, maybe relative to Macro / Mod / Config / document, where to look for function code
- list of function files

I'd rather not provide for larger pieces of Python code inside an FCStd file.

Python is still quite new for me, too. 
So, I'd appreciate any hint from the security specialists in these questions.

FlachyJoe, 2024-06-02:
About security: could we prevent user-launched python to modify files out of the Mod and Macro directories? Or alert the user if it tries to change such file without shown a save dialog before?

wolfgangr, 2024-06-02:
> could we prevent user-launched python to modify files out of the Mod and Macro directories?

well, yes, that's part of the basic idea behind sandboxing.

I'd go even further: 
The Idea behind **user defined spread sheet functions** is **informative, not manipulative.**
So, I'd rather **not allow to modify anything** at all - not even FreeCAD objects in the same file.
I can **produce output** by writing a result into a `App::PropertyPythonObject` Property,
(optionally) postprocess this in my spreadsheet, and then put an expression in any model **feature** property to **PULL** the result, as it is the common FreeCAD way described here:
https://wiki.freecad.org/Manual:Using_spreadsheets#Writing_properties

The question is, how to achieve this level of control without missing the easy way of 'evil `eval`'.
I'm just searching the web for "Python eval sandbox" to avoid reinventing the wheel.

https://docs.python.org/3.8/library/ast.html#ast.literal_eval
very restrictive, cannot even parse simple arithmetic expressions like "3*4"
I think I will try this one to parse my paramaters
https://github.com/wolfgangr/myTinyAsm/blob/41b00bbe46b4101b7f76a68de0d1874adc2c5b1f/sheetPythonCustomFunc.py#L87
since some lines earlier I try to `obj.evalExpression` for Parameters like `=3*4` i.e. we can use FreeCAD expression engine for quite a lot of input cases.

For function body itself, I consider

- check for alnum, but I think this leaves a lot of backdoors open
- https://lmfit.github.io/asteval/
which says is designed 'evaluation of mathematical expressions' - which may come close to our case
- https://github.com/yaroslaff/evalidate/blob/master/README.md
which may offer more speed and option, but looks harder to understand
- https://github.com/danthedeckie/simpleeval
still to be checked
- https://gist.github.com/miohtama/34a83d870a14aa7e580d aka safeeval

https://github.com/vstinner/pysandbox/
> pysandbox is BROKEN BY DESIGN, please move to a new sandboxing  !!!
> !!! solution: run python in a sandbox, not the opposite! 

I'm afraid that this warning refers to FreeCAD as a whole as well.
You can't teach your kiddie to operate a chain saw without any access to one.
Think the design teams policy "keep code separate from data" is skd of compromise:
Make access to code hard enough so users don't get trapped by mistake or by malicious code in some nice looking FCStd file.
However, everybody can throw malicious code from whatever source in his Macro directory - as kiddy can grab into a running saw chain.

So my current idea for a security concept looks like this:

- pass only strings, FreeCAD-Expressions and python-literals passing ast.literal_eval as arguments
- check function name by something restrictive as asteval

wolfgangr, 2024-06-02:
after some recherche, I decided to start with `evaildate` sanity checker:
https://github.com/yaroslaff/evalidate/

- I expect it to do what I want
- it implements a strict "forbidden if not explicitly allowed" policy
- compared to other tools, it filters at the run time, not the compile time stage
so any user supplied code is restricted as well
- it promises to be fast
- it's just 135 lines of code are easy to understand and audit
- it is distributed as pip, so at least subject to "many thousand eyes control"
- it's MIT-License appears compatible to FreeCAD, as I hope
- ... and allows us do distribute it in copy, restricting the risk of malicious pip source takeover

wolfgangr, 2024-06-02:
this was the plan:
> So my current idea for a security concept looks like this:
> - pass only strings, `FreeCAD`-Expressions and python-literals passing `ast.literal_eval` as arguments
> - check function name by something restrictive as asteval

The first step is implemented in verbatim now:

https://github.com/wolfgangr/myTinyAsm/blob/33855a6d78286e84c7e2cb3545af1165a3b7f7f4/sheetPythonCustomFunc.py#L63C1-L101C24
```
# evaluate parameters from def-List
def eval_param(obj, param: str):
    # is it a string?
    match = re.search(r"^\'(.*)\'$", ps )
[ ... ]
    # '=...' try  FreeCAD Expression;  even matches ...
    #   <<strings>>, cell and object references,  numbers and arithmetic expressions
    match = re.search(r"=(.*)", ps )
[ ... ]
    if match:
        try:
            evld = obj.evalExpression( match.group(1) )
        except:
            return None

    # try python to eval it:
    try:
        return literal_eval(ps)
    except:
        return None
```

For the function part, `evalidate` did not match my initial expectation to deliver sandboxing for the user supplied code as well. I had to realize that the scope of `evalidate` ends at securing the translation the function name to its representing object.
However, by properly declaring the allowed functions in `model=sPyMod_model`, we might (so I hope...) at least be sure that no malicious code is injected from the data area - in compliance with current `FreeCAD` policy.

https://github.com/wolfgangr/myTinyAsm/blob/33855a6d78286e84c7e2cb3545af1165a3b7f7f4/sheetPythonCustomFunc.py#L125C1-L127C27

```
    func_evd = evalidate.Expr(funcnam, model=sPyMod_model)
    func_ptr = func_evd.eval()
    rv = func_ptr(*params)
```

This extends the configurable properties in the sheet to three:

- list of search pathes
- list of file names
- list of allowed functions

Thus, code itself is still kept in seperate (`*.py`) files, outside `FCStd` files, to comply with `FreeCAD` policies.

I'd plan to autogenerate the Dictionary of allowed functions for `evalidate` automagically from this configuration infos.
This way, the 'bomb' may reside in a (`*.py`)-File, but the complete 'trigger' is in the `FCStd`.

I'd like the possibility of keeping both in the same directory, so they can maintained, backuped, distributed, versionized, understood, audited etc. coherently. I feel keeping them e.g. in the `Macro` directory might add a level of obscurity, but that may be simply a personal view of some `FreeCAD` novice, of course.

From my point of view, the current level is much more of security I ever intended to implement for my own purpose.
Of course, if there is some desire in the community to pick up my ideas, from this starting point, there may be lot of space for more perfection.

**To summarize:**

- **Arguments** are safe by limiting to pure strings, `FreeCAD`-Expressions or `ast.literal_eval`
- **Function name** is secured by `evalidate`
- **Function content** in separate `*.py` file is as open as in every Macro, as accepted by the current `FreeCAD` policy

wolfgangr, 2024-06-05:
While trying to implement my plan, I found that by preparing for '`evalidate`', we make it obsolete for our case:
In the process of a controlled interpretation on sanitized module path, we get a dictionary {function_name: function_reference} anyway. 
This suffices for calling, no need for parsing, so no chance for malcious code being injected to any parser.

Details available here:
https://github.com/wolfgangr/myTinyAsm/blob/5e604294af707c9d6e9cb269d7d5711515aa4e6c/spEvalidate.py#L59

wolfgangr, 2024-06-05:
Screenshots and some first description of the current state is available in the forum:
https://forum.freecad.org/viewtopic.php?p=764197

onekk, 2024-06-13:
Asking for new user defined functions is a good thing.

injecting python code somewhere IMHO is not a good decision, as now you should load and install a macro or an addon explicitly, adding the ability to add auto executing code in the FCStd file without the user being advertised is a no go. This is the main security concern. An user could intentionally damage his system.

The point is that opening a file that is executing any type of code that could alter the system the program is running on is bad and should be avoided.

This is the security problem.

Sandboxing could be a solution, but you can't expect that each program has to setup a "securized virtual environment".

Adding another script language is even worse as you have to maintain and care about two codebases and you will duplicate deceloped efforts.  Python is used in many scientific environments do it should be not too bad as language.

However your work is interesting, but I can't guess the final goal, as it seems more a another way to do assembies.

wolfgangr, 2024-06-18:
> injecting python code somewhere IMHO is not a good decision,

agree.
That's why - contrary to my first ideas - I decided only to accept code that resides in the Macro directory, not in the FCStd file path, somewhere else or even in the FCStd.

So, in my current version, the user has explicitly to

- install the FPO-code for the pythonized spreadsheet where FreeCAD finds it
- install his (or someone else's) user callable python functions in the Macro directory

Only after this intentional user actions, additional code is available.

> I can't guess the final goal, as it seems more a another way to do assemblies.

That's correct, as the folder name indicates:
A lean, minimalistic, datum based, pure Python assembly toolkit is the larger motivation behind the collection I developed together.
I just finished the last point on my to-do-list - a tiny animator. So all the bits are in skd of "proof of concept" stage.
When I managed to put the bits together in a workbench, I'll present that in the forum.
I don't think that the github issue here at hand is the correct place to discuss the wherabouts of "just another assembly"

Anyway, the spreadsheet functions are independent of the other code bits living in the same github folder.