FreeCAD/FreeCAD #14044: Pull Request #12929 introduced regression in SpaceMouse configuation for FreeCAD

andreymrt, 2024-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Pasting from https://github.com/FreeCAD/FreeCAD/pull/12929#issuecomment-2113534107
-----------

It looks like PR #12929 introduced a regression where it is no longer possible to configure a spacemouse for FreeCAD
Relying on the 3DX config window as suggested in the comments of #13801 by @pskowronskiTDx.
When setting Zoom direction to "Up/Down", it does not result in the expected behavior.
Instead of the model zooming in/out, it just pans in screen Y axis as if nothing changed. Deactivating axes in the config window also has no effect.
Restarting FreeCAD between changing 3DX settings has no effect.
Restarting 3DX has no effect.

Screen cap:

https://github.com/FreeCAD/FreeCAD/assets/1222366/74856d7d-fe31-46d1-af27-678e9af1a7ca

Can someone confirm whether or not you experience the same behavior in the latest build?

To further check driver behavior, when using KiCad 8.0.2, 3DX works fine.
And deactivating axes in the config window does indeed limit the movements affected in the 3D viewer.
So I think we can rule out an issue with the 3DX driver.

### Full version info

```shell
FreeCAD

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:
  * Help 1.0.3
```

3DxWare
```
3DxWare 10.8.20.3729
3DxWinCore 17.8.20.20889
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-16:
is this related to https://github.com/FreeCAD/FreeCAD/issues/13801

andreymrt, 2024-05-16:
That's what I thought at first, but
- I'm running a new spacemouse
- Driver is updated to the latest version
- FreeCAD is still getting inputs and controls from 3DX
- Other programs work fine
- Older FreeCAD works fine
- This [comment in #13801](https://github.com/FreeCAD/FreeCAD/issues/13801#issuecomment-2111750449) states that the functionality that used to be in the old Spaceball tab page is now supposed to be in the 3DX driver config app, as I'm using it in the attached screencap

@maxwxyz do you have a SpaceMouse and can you confirm how things work for you with latest nightly?
What else can I do/provide to help diagnose this issue?

maxwxyz, 2024-05-16:
Unfortunately no, I hope  @pskowronskiTDx can weigh in on this.

pskowronskiTDx, 2024-05-16:
@andreymrt Having the exact same verisons of software, I was unable to recreate the issue on my end so far. I keep looking into this...

maxwxyz, 2024-05-16:
@Roy-043 could you add this in the wiki?
I'll close this issue then.

closed 2024-05-16 by maxwxyz