FreeCAD/FreeCAD #14049: Sketcher: Dimension arrows missing

maxwxyz, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the current weekly either all or one dimension arrow is missing from the constraints:
[image]

Not sure which PR caused this regression https://github.com/FreeCAD/FreeCAD/pull/13966, https://github.com/FreeCAD/FreeCAD/pull/13844, ...?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * fasteners 0.5.20
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-16:
@FlachyJoe @PaddleStroke @bgbsww  FYI

FlachyJoe, 2024-05-16:
Found, a PR is coming.

furgo16, 2024-05-17:
I can confirm. I noticed this a couple of days ago, and I thought that that was a design decision. Glad to hear it's a bug.

maxwxyz, 2024-05-17:
@FlachyJoe is this related to:
- https://github.com/FreeCAD/FreeCAD/issues/13557
The arrow tips also seem to change side when viewing from behind.
Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/5fc2c8ca-2be9-4e43-a6a9-0ac0906e2722

PR #14053 (2024-05-20): Gui: fix SoDatumLabel arrow and text disappear
Fix #14049 
Fix #13557
arrow drawn CCW and text was not shown in every cases.
closed 2024-05-20 by chennes