FreeCAD/FreeCAD #14051: TechDraw dimemsion repair inconsistent selection with greedy selection

johnsonm, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using TechDraw_DimensionRepair, after clicking "OK" to apply a repair, the repaired dimension is selected in the tree, but not in the drawing (it is no longer green to indicate that it is selected).

With greedy selection, and multiple dimensions to repair, I then click on another dimension to select it, click to repair, and select the correct references, click "OK", and the *previous* dimension is what was actually being modified and is then updated.

Even though nothing is showing as selected in the document (nothing is green), I still have to *first* click the background to deselect the dimension in the model tree, then click on the dimension to repair, in order to actually repair it.

For example, loading https://gitlab.com/mcdanlj/RotaryBroach/-/raw/main/RotaryBroachParts.FCStd?ref_type=heads&inline=false and repairing references on BodyBoreOpsPage, I can have (as here) have just repaired Dimension007, then select Dimension001 on the page, silently leaving both Dimension0001 and Dimension007 selected in the tree, and Dimension007 the one actually chosen to repair. If I don't notice that, then when I replace the references, the wrong dimension changes.

[image]

When finishing a repair, either de-selecting the repaired dimension in the tree or re-synchronizing the view in the page to the selections would avoid this problem. It would be better to de-select the repaired dimension in the tree because if TechDraw_DimensionRepair is required at all, it is probably required for multiple dimensions, therefore the workflow is probably repairing multiple dimensions in a row, and having to explicitly deselect is additional and unnecessary work.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37308 (Git)
Build type: Unknown
Branch: main
Hash: b72a8c4222e74205fcbc181efc2d3182e8c59da5
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-05-26:
I agree, clearing the selection on exit works better. Thank you!

Could dimension repair also honor the greedy selection configuration? I note that in dimension repair I still have to control-click even though I have greedy selection enabled for TechDraw.

johnsonm, 2024-06-03:
I think the underlying problem isn't actually repair. I just tested now with 77fc1de6 and realized that with greedy selection, the tree view doesn't match the selected view. That wasn't obvious to me until the other fixes landed. ☺

Starting with nothing selected:

[image]

Click on Dimension001 on the page (not in the tree), so far so good:

[image]

Click on Dimension002 on the page (again, not in the tree), now the tree selection doesn't match the visible selection on the page:

[image]

So only Dimension002 is green on the page, so if I were to do anything with what looks there like the one selected dimension (for example, repairing it, though it no longer needs repair thanks to other fixes), the first selected dimension in the tree will be selected (here, Dimension001). 

If I don't catch that the selected dimension isn't what I thought I was working with by having happened to have noticed names, my changes apply to the wrong dimension and it's very confusing.

So it would be nice if with greedy selection of dimensions, the rendered view on the page matched the selections in the tree. Here, I'd want both Dimension001 and Dimension002 to be green on the page and selected in the tree, so that they are in sync. (Obviously, if I'm repairing, I realize that's wrong, click background to reset the selection, and start over.)

johnsonm, 2024-06-04:
Based on fixes to other bugs clarifying what was really happening, I have updated the description of this issue to reflect what I currently understand to be the underlying issue.

WandererFan, 2024-06-17:
Should be fixed by  #14737

johnsonm, 2024-06-18:
I confirm fixed in 2ca9b6ef8036 — thank you!

closed 2024-06-18 by johnsonm