FreeCAD/FreeCAD #14055: Missing things at dxf import

hw-schrauber, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I import the attached dxf only the outline contour is imported. The two inside objects are not imported. A work a round is to open the dxf in a different cad program, export to dwg and import the dwg in FreeCAD. Dxf-direct-import shows 121 items. Via dwg the same shows 171 items.
[GB 76.zip](https://github.com/FreeCAD/FreeCAD/files/15335115/GB.76.zip)

### Full version info

```shell
[GB 76.zip](https://github.com/FreeCAD/FreeCAD/files/15335115/GB.76.zip)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-16:
Please provide full about info of the version you are using. Did you try this in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-07-25:
@hw-schrauber please add full [About](https://wiki.freecad.org/About) info

yorikvanhavre, 2024-07-30:
Looks ok to me. Can you try a development build?

[image]

luzpaz, 2024-07-30:
No response from OP. Closing. Please re-open if develpment build shows incorrect behavior.

closed 2024-07-30 by luzpaz