FreeCAD/FreeCAD #14056: Assembly: Problem with CopyOnChange links

wifasoi, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I noticed this bug wen creating some assembly with generic extruded aluminum profiles (implemented with the "CopyOnChange" feature of App::Link). Wen you modify the copyOnwrite parameter, the solid will shift, and if you try to solve the constraint, the solid will shift more, and if you try to modify the parameter, will shift again, in a random rotation and magnitude.

A possible way to reproduce the bug:
- Creating an assembly;
- Add a part as a link, and ground it;
- Add another part as a link (that has a CopyOnChange parameter that modifies the solid), and set Link On Copy on "Enabled"
- make a "Fixed joint" constraint between the two object
- Change the CopyOnChange parameter
(The solid now swifts randomly)
- Press "Solve Assembly"
(The solid will shift more each time is pressed)

[image]
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Europe (en_150)
Installed mods: 
  * Assembly4 0.50.12
  * Gridfinity 0.4.0
  * kicadStepUpMod 11.1.0
  * LCInterlocking
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Assembly

### Anything else?

I tried with different 
No error log are generated in the console, But i created an example that triggers the strange behavior.
[testbug.zip](https://github.com/FreeCAD/FreeCAD/files/15335026/testbug.zip)

You just need to press "Z" (Solve Assembly) to admire the majestic glider going palces :D

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-17:
Ah yes I have not taken this into account so it is undefined.

FEA-eng, 2026-09-17:
```
07:24:40  <Expression> ObjectIdentifier.cpp(817): duplicate object label testbug#Profile
07:24:40  Assembly joint 'Assembly.Fixed' has an invalid 'Object2' or related attributes. 'NoneType' object is not subscriptable
07:24:40  Assembly joint 'Assembly.Fixed' has an invalid 'Reference2' or related attributes. 'NoneType' object is not subscriptable
07:24:41  <Exception> ObjectIdentifier.cpp(1463): Property 'Profile' not found in 'Box'
in property binding 'testbug#Box.Length'
07:24:41  Box: Property 'Profile' not found in 'Box'
in property binding 'testbug#Box.Length'
```

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```

maxwxyz, 2026-09-17:
closing then if no other file is provided

closed 2026-09-17 by maxwxyz