FreeCAD/FreeCAD #14058: Missing status bar (KDE 6.02) on 0.21.2

kaiwas, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I lost the information panel after updating KDE to version 6.02
Lost in version 0.21.2. In version 0.22dev everything is fine.

I tried resetting the settings and it didn't help.
[image]
[image]

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-16:
I guess if it's already fixed in the weekly builds this can be closed.

kaiwas, 2024-05-16:
> I guess if it's already fixed in the weekly builds this can be closed.

Yes. in Version 0.22 everything is fine.
But perhaps someone will use version 0.21 even after the release of 1.0

My job is to inform. The developers know better what to do with it ;)