FreeCAD/FreeCAD #14061: Draft commands on a Techdraw page breaks FreeCAD

krushia, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. File > New
2. View > Workbench > TechDraw
3. TechDraw > Page > Insert Default Page
4. View > Workbench > Draft
5. Drafting > Line

```
14:44:43  Running the Python command 'Draft_Line' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCA8kdne8/usr/Mod/Draft/draftguitools/gui_lines.py", line 82, in Activated
    self.call = self.view.addEventCallback("SoEvent", self.action)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^

'NoneType' object has no attribute 'addEventCallback'
```

From this point, something is "stuck" and any action will result in the error `<class 'AttributeError'>: 'NoneType' object has no attribute 'removeEventCallback'`. Restarting FreeCAD appears to be the only option.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-16:
Doesn't break in FC 21.2

krushia, 2024-05-16:
Same problem when using Draft commands with a Spreadsheet tab selected. Also with a Text document tab. Sketcher tab will disable the Draft menus.

maxwxyz, 2024-05-16:
@Roy-043 fyi

Roy-043, 2024-05-16:
This is sort of a known issue. The Draft commands should be disabled in such cases.

Related to #10634.

yorikvanhavre, 2024-05-17:
Or, some Draft tools might at some point be made to work in TechDraw :sunglasses:

Roy-043, 2024-05-17:
I assume some <s>Arch</s> BIM commands have the same issue. I have a PR prepared for the Draft WB, but I will leave this open until the BIM WB is checked as well.

yorikvanhavre, 2024-05-22:
> I assume some Arch BIM commands have the same issue. I have a PR prepared for the Draft WB, but I will leave this open until the BIM WB is checked as well.

Working on it!

PR #14207 (2024-05-27): BIM: improved IsActive behaviour
fixes #14061
commit a9e40565 (2024-05-27): BIM: improved IsActive behaviour - fixes #14061
closed 2024-05-27 by chennes