FreeCAD/FreeCAD #14062: colors keep grey when changing colors

diepchess, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

if i try change colors of something in the treeview it doesn't show at the screen.
Action performed: i click in treeview onto for example some object. I go to combo view there into the 'view' tab go to shape color. click on it. a popup is there with 'select color'. i click a different color. click ok and still the union of objects displays as grey at screen. Version 21.2 doesn't have this bug with the same file. Not this file is alucncFCLB20240407x047 or 048 - if i select the union of objects and say 'copy'. Then open new document. put the union in that object, then the bug isn't there and it is coloring it correctly. So it seems happen when there is multiple unions or i call it objects in the treeview. Seems the 26+ MB the 7-zip makes of the volumes doesn't post well here. please mail me at diep at xs4all dot nl to email it maybe someone can attach the file then. 
[image]

-rw-r--r-- 1 diep diep 36141472 May 16 18:22 alucnc2024FCLB20240407x048.FCStd
-rw-r--r-- 1 diep diep  1341496 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.006
-rw-r--r-- 1 diep diep  5000000 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.002
-rw-r--r-- 1 diep diep  5000000 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.003
-rw-r--r-- 1 diep diep  5000000 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.004
-rw-r--r-- 1 diep diep  5000000 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.005
-rw-r--r-- 1 diep diep  5000000 May 16 18:22 alucnc2024FCLB20240407x048packed.7z.001

### Full version info

```shell
[code] 
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 2024.408.0.1455 (Git shallow) 
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-16:
Your image shows you using realthunder's Link branch. Can you reproduce it with vanilla FreeCAD?
Link branch issues go here: https://github.com/realthunder/FreeCAD/issues

diepchess, 2024-05-16:
Yes i tested weekbuild 0.22.0dev rev 37302 from 5 may 2024
and it doesn't have the same issue.

seems linkbranch only unlike what Chris Kaminski told me initially.

diepchess, 2024-05-16:
Yes sorry posted here wrong. Doesn't post at realthunder issue tracker somehow. Weeklybuild 0.22.0dev from 13 may 2024 doesn't have the same issue!!

diepchess, 2024-05-16:
Yes sorry posted here wrong. Doesn't post at realthunder issue tracker somehow. Weeklybuild 0.22.0dev from 13 may 2024 doesn't have the same issue!

closed 2024-05-17 by maxwxyz