FreeCAD/FreeCAD #14064: Visibility icon does not work with Link Elements

NomAnor, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Link elements are shown in the tree, clicking on the corresponding visibility icon will not toggle their visibility. Using `Space` works and the icons show the items as hidden.

Clicking on the icon of the link itself works.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36534 +178 (Git)
Build type: Debug
Branch: integration
Hash: b87e0ddfb5858c867abaecca3b3134ef2ababfe7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

Core

### Anything else?

The code uses the `hide()` and `show()` methods of `ViewProviderDocumentObject`
https://github.com/FreeCAD/FreeCAD/blob/acecc2c4356b064b0e6abec7cce305b6508e3845/src/Gui/Tree.cpp#L1623-L1633

https://github.com/FreeCAD/FreeCAD/blob/acecc2c4356b064b0e6abec7cce305b6508e3845/src/Gui/ViewProviderDocumentObject.h#L72-L75

I'm not sure if this is a bug in the `hide()` and `show()` methods or if the code should use some other methods.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
@PaddleStroke I see you experiment a lit with links in assembly, can you confirm this?

PaddleStroke, 2024-05-17:
You need to check how the 'toggle visibility' command (the command triggered by space) works and if it handles links specifically.
Then just replicate the correct way of doing with the eye icon.

maxwxyz, 2024-05-17:
@NomAnor are you tackling this, as the original PR was yours (https://github.com/FreeCAD/FreeCAD/pull/12298)?

kadet1090, 2024-05-17:
> You need to check how the 'toggle visibility' command (the command triggered by space) works and if it handles links specifically. Then just replicate the correct way of doing with the eye icon.

The icon should not replicate the correct way but simply properly trigger that command. This is the only way to ensure that the behavior will be consistent.

NomAnor, 2024-05-17:
I was trying to do that,  but if I remember corretly, the command was written such that it used the current selection. But I needed some way to show/hide an arbitrary object. The mentioned methods looked like they would do that.

I will try to look into the code. If possible, I think the toggle visibility command should be implemented in terms of some `show(object)` and `hide(object)` functions. That would prevent inconsistencies between the implementations.

kadet1090, 2024-05-17:
So it should be refactored in a way that toggling code is common between them
.i.e. eye icon toggles the item that it is associated with and action toggles selection

NomAnor, 2024-05-19:
I looked into the code and I found the difference. I don't understand it yet.
This is the code that toggles visibility for the `Space` command:
https://github.com/FreeCAD/FreeCAD/blob/9decb98c71fe7b3299d422c2b0b92f0507c11c91/src/Gui/Selection.cpp#L1296-L1357

This is the code the visibility icon uses:
https://github.com/FreeCAD/FreeCAD/blob/9decb98c71fe7b3299d422c2b0b92f0507c11c91/src/Gui/ViewProviderDocumentObject.cpp#L283-L300
https://github.com/FreeCAD/FreeCAD/blob/9decb98c71fe7b3299d422c2b0b92f0507c11c91/src/Gui/ViewProviderDocumentObject.cpp#L225-L234

The selection code uses the `setElementVisible` API from the parent and only falls back to the `ViewProviderDocumentObject` if there is no parent.
The `ViewProviderDocumentObject` code essentially changes the `Visibility` property but that seems to be unchangeable for `LinkElement`.

I don't know how the sub-element stuff works. The `setElementVisible` API looks to me more generic. I might be able to refactor the logic from the selection code into it's own helper function that can be used with a list of objects without requiring a selection.
This can then be used by the selection code and the visibility icon.

Is there a place for such standalone helper functions?

NomAnor, 2024-05-28:
I created a draft PR that seems to fix it, but I'm not sure if the code is good.

PR #14341 (2024-06-03): Fix visibility icon for link elements
The visibility of link elements must be set with setElementVisible. This will first try to use that API otherwise it falls back to setting the Visibility property.

This seems to work, but I'm not sure if this is correct. I don't understand when to use `setElementVisible` and what the interactions are with sub elements and such. 

Fixes #14064
closed 2024-06-03 by chennes