FreeCAD/FreeCAD #14066: Wrong tree icons for PD Pad and Revolution

NomAnor, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The icons for Pad and Revolution are blue, but the toolbar icon are yellow. Loft for example uses the same icon for the toolbar as for the tree. Pad and Revolution should do the same to be consistent in the color scheme for additive and subtractive features.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git)
Build type: Debug
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

There are two special icons:
* [Tree_PartDesign_Pad.svg](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/PartDesign/Gui/Resources/icons/Tree_PartDesign_Pad.svg)
* [Tree_PartDesign_Revolution.svg](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/PartDesign/Gui/Resources/icons/Tree_PartDesign_Revolution.svg)

while the others features all seem to use the `PartDesign_*.svg` icons.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
not sure why but this was done intentionally at some time. Same in Part WB. All tree icons should be blue was the intention.
@FreeCAD/design-working-group

kadet1090, 2024-05-17:
Frankly, I don't see a good reason why it should be all blue. It is harder to quickly connect the action on toolbar with feature in the tree. Additive/Subtractive features are also harder to distinguish that way.

Coloring them all blue would make more sense If we do additive/subtractive like it done in other cad packages so it is the same feature with simple switch to make it subtractive or creates new body instead. Not saying that it would be better design but at least it would be easier to implement.

maxwxyz, 2024-05-17:
I agree but then all should be updated, not just 2

kadet1090, 2024-05-17:
Description says that only those two are special cases - so I assume that other features already reuse icon from toolbar. if that is not the case - sure we should make it consistent at least in part design and part workbenches

NomAnor, 2024-05-17:
> Frankly, I don't see a good reason why it should be all blue. It is harder to quickly connect the action on toolbar with feature in the tree. Additive/Subtractive features are also harder to distinguish that way.

That's what I'm thinking as well. 

I would keep this issue confined to Part and PartDesign. For other workbenches new issues should be created if there are also inconsistencies.

I looked through PartDesign and so far I see the following:
* Pad has a different icons for the tree (blue) and the command (yellow)
* Revolution has a different icons for the tree (blue) and the command (yellow)
* InvoluteGear and InternalExternalGear are the same icon the first seems not to be used
* Body (command) and Body_Tree (tree) are the same icons
* Body_old seems to be unused

These are easy to resolve. 

I will look into Part later.

maxwxyz, 2024-05-17:
If possible, it should be consistent, Part WB as well.

Not sure about BIM/Arch: most commands are yellow and tree icons white
@yorikvanhavre what's your opinion on that and do you know the history of the current implementation?
@marcuspollio FYI

obelisk79, 2024-05-17:
Agree with just making the tree icons consistent with the toolbar icons. special cases don't seem to make any sense, perhaps just leftovers from long ago?

NomAnor, 2024-05-18:
Here are my findings from the Part Workbench:
* The toolbar primitives have different icons for the tree (blue, `parametric` folder) and the command (yellow, `create` folder)
* The primitives from the "Create primitives" command have blue icons
* The features/tools use mostly blue icons

For Part it seems that the toolbar primitives are the outlier, everything else ist blue. 

### Part
I'm not familiar with the Part workbench, but the same color theme for all features make sense. As far as I can tell all features create some kind of new oder altered geometry and this new geometry seem to be represented in blue. Whereas the support geometry is depicted in red. It's not always the case, Part extrude has no red "Wire" vs Part Desing extrude (tree) for example. Booleans are the only way to remove something form the geometry.
Sweep and Loft look the same as in Part Desing, just the colors are inverted. This could make it more difficult to distinguish between Part and Part Design Features, but Part Design features are always inside a Body.

Based on this the primitives would be better to use their tree icons (showing their parameters in red) than the current yellow ones.
Some icons could be altered to fit in with the red support theme (Extrude, revolve, fillet, chamfer).

### Part Design
In PartDesign there is a clear distinction between additive and subtractive features. Currently distinguished in the toolbar by a yellow or blue/red icon. Red on the other hand is not used consistently, in the additive features is seems to depict the support geometry and in the subtractive features the tool shape that gets removed.
Then there are altering features (fillet, chamfer, draft, thickness), here blue is used for the existing geometry and red for the altered faces.
The transformation features show the existing geometry in blue and the new patterned geometry in yellow. If the new geometry is additive or subtractive depends on the selected features.

Using the toolbar icons in the tree for the additive features feels to be more in line with the rest. And it would help to distinguish them in the tree.

### 1.0
I don't think the icons are bad and I'm probably used to them so they don't seem confusing. Looking at them to figure out what the colors could mean was interesting. They are somewhat inconsistent and I can understand that a new user could be confused.

If 1.0 should benefit from a bit more consistency in the icons, I would propose the following changes:
* Change Part primitive toolbar icons to use their tree icons
* Change Part Design tree icons to use their toolbar icons
* Make the support wire in Part Design Pad red instead of white.

Then Part icons are predominently blue and consistent in the tree. Part Design icons are also consistent and a bit better to distinguish. Both changes are easy to make and hopefully not to disruptive for existing users.

For the future the DWG could discuss if a more stringent use of certain colors (e.g. support geometry) is needed to help new users understand the features based on the icons. We should create a new issue for this and close this one if the above changes are merged.

maxwxyz, 2024-05-18:
I agree, thanks for the thorough look into this!
@FreeCAD/design-working-group FYI

NomAnor, 2024-05-18:
I added all changes to my draft PR. Here are some screenshots:
[image] [image]

[image]
[image]

[image] 
[image]

[image]
[image]

Someone more talented than me should look at these three icons:
* PartDesign_Pad (The black outline is not consistent with the darker red outlines of the other icons)
* create/Part_Primitives.svg (should be blue)
* create/Part_Shapebuilder.svg (should be blue)

maxwxyz, 2024-05-18:
@Reqrefusion maybe? you can post the .svg files directly here in a comment

Reqrefusion, 2024-05-18:
@NomAnor Very nice work, congratulations. While you're trying to create consistency in colors, I'm trying to do it in size. When @maxwxyz  said svg file, I think he wanted me to throw away the keyline I used.

All icons follow this pattern according to their shapes and occupy equal space on the plane, so none of them stand out or fall behind. I'm working on adding this to the Guide. It is not in the Guide right now, but if you follow it, I would be very grateful.

The relevant svg file is here. Since it is drawn in black, it does not appear on a dark background. I would be happy if you download it and check it out.
///////////////
[image]////
///////////////

maxwxyz, 2024-05-18:
@Reqrefusion no I meant maybe you can help with the last 3 icons he mentioned and post them here.

Reqrefusion, 2024-05-18:
> @Reqrefusion no I meant maybe you can help with the last 3 icons he mentioned and post them here.

Oh sorry. My mind is always there, unfortunately, of course I want to help. It would be especially good to put an end to the ugliness of the part primative. Of course, it would be healthier to see your opinion first. After all, you are the captain of this transformation, @NomAnor .

Reqrefusion, 2024-05-18:
[image]
[image]
[image]
Done. I couldn't help but solve a few more problems, but I held back for sizing. I made it according to your comment @NomAnor , so there may be something missing, please tell me.

MisterMakerNL, 2024-05-18:
Why is the extrude floating above the sketch tho?

Reqrefusion, 2024-05-18:
> Why is the extrude floating above the sketch tho?

I don't know, you have to ask the person who did it first. Since this is just a work to make the colors more harmonious, the absence of renewals is normal, renewals are now for other PRs. I also really want to change part_primative.

MisterMakerNL, 2024-05-18:
> > Why is the extrude floating above the sketch tho?
> 
> I don't know, you have to ask the person who did it first. Since this is just a work to make the colors more harmonious, the absence of renewals is normal, renewals are now for other PRs. I also really want to change part_primative.

Owh if you feel ambitious make another PR and fix those weird things that makes no sense 👍 
Like these 2:
[image]
[image]

Reqrefusion, 2024-05-18:
> > > Why is the extrude floating above the sketch tho?
> > 
> > 
> > I don't know, you have to ask the person who did it first. Since this is just a work to make the colors more harmonious, the absence of renewals is normal, renewals are now for other PRs. I also really want to change part_primative.
> 
> Owh if you feel ambitious make another PR and fix those weird things that makes no sense 👍 Like these 2: ![Tree_PartDesign_Revolution](https://private-user-images.githubusercontent.com/29804962/331831790-9d4ad089-e463-468b-84d4-424d648538a6.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYwNjYyODYsIm5iZiI6MTcxNjA2NTk4NiwicGF0aCI6Ii8yOTgwNDk2Mi8zMzE4MzE3OTAtOWQ0YWQwODktZTQ2My00NjhiLTg0ZDQtNDI0ZDY0ODUzOGE2LnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE4VDIwNTk0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkOGRiMjBjNmM0NTMzNmJjOWUzNmY1Y2U4NDc4NGM1NjI2ZGMwMzJlZjE2NWMxNGIxYTUwYTI0ZGJjYWI4ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EfL82kad6Ley0IR65UUTGQDpPKUev4dVuCkCCPMUXPs) ![Tree_PartDesign_Pad](https://private-user-images.githubusercontent.com/29804962/331831791-48da71cf-0b7d-4f59-97e7-b854cc7a86c3.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYwNjYyODYsIm5iZiI6MTcxNjA2NTk4NiwicGF0aCI6Ii8yOTgwNDk2Mi8zMzE4MzE3OTEtNDhkYTcxY2YtMGI3ZC00ZjU5LTk3ZTctYjg1NGNjN2E4NmMzLnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE4VDIwNTk0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBhYTY0MjI4NjViM2EyZDU3ZGM5YTg3MTc1NGJmNzA2MGM2NmE5YmQxNGY5NWFhZDlkMjk2MzQ4ZTZiMTkzNjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0qSu8KX5z_TncuuYMUd0Ti-FcsZYJtHFnsh2QTEmFNY)

I'm going over each icon one by one, and these will come from time to time. There are a lot of flaws that I haven't touched even when correcting the colors. I even thought one of them was a feature. Open a topic on discord for these so I don't forget when the time comes.

NomAnor, 2024-05-19:
> The relevant svg file is here. Since it is drawn in black, it does not appear on a dark background. I would be happy if you download it and check it out.

I lack the skill to create icons, so your help is appreciated.

[image]
[image]

I added Reqrefusion's icons and removed some unused files. As far as I can tell, all PartDesign tree icons are the same as the toolbar icons and all Part icons have a blue theme.

I update my PR for review and if there are no objections, it's merge can close this issue.

@Reqrefusion do you want to open a new issue to discuss and track the overhaul of Part and PartDesign icons according to the newer design guide?

Reqrefusion, 2024-05-19:
@NomAnor Great congratulations. I personally struggle with this sizing. I started a discussion to write a new guide for this to be added to the guide. It will probably be in the new guide, but I don't think it's solid enough to become an issue.

PR #14074 (2024-05-20): Change PartDesign Pad and Revolution tree icons
Use the same icons as the toolbar buttons.

Fixes #14066.
closed 2024-05-20 by chennes