FreeCAD/FreeCAD #14067: BUG: Sketcher grid not changing state between ON and OFF

JDMils, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With default settings applied, opening FCAD and going to Sketcher shows the XY axis with the grid in an off-white colour. I want to turn the grid off so I click:
Edit>Preferences>Sketcher, open Grid tab and uncheck the Grid checkbox. Click Apply then OK. Grid is still visible. 
I can now close the Sketcher and re-open it and the grid still displays. I have to close the current document and create a new one in order for the grid to not show on screen.
Same thing when I try to enable the on-screen Grid.
If I open a new sketch in the current document, the new grid settings are applied. In this case, I will have at least 2 sketches in my document, and opening the first one will show the grid while opening the 2nd sketch will not show the grid. Looks like the grid setting are stored in the sketch.
I saved the document with the 2 sketches and re-opened FCAD and the document and the grid settings are still tied to the sketch. Opening the first sketch shows the grid while opening the 2nd sketch does not.

https://github.com/FreeCAD/FreeCAD/assets/26943601/ec63e1cd-725e-459a-8dff-afd0c4b9b23b

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
It should be solved already, can you test this please in the current dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roy-043, 2024-05-17:
There is a misunderstanding here (not a bug and nothing has been solved AFAIK). See forum topic. This should be closed IMO.
https://forum.freecad.org/viewtopic.php?t=87702

closed 2024-05-17 by maxwxyz