FreeCAD/FreeCAD #14068: Fasteners not visible on arrays

iulyus2004, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello!

As I was trying to add some washers, bolts into my assembly, I have a problem related to the visibility of the fasteners which are attached to the arrayed items,   I am able to see the fasteners into the main tree, but they are not visible even if I press space bar. This problem persists just on the items where I used array option.

[image]

### Full version info

```shell
OS: Linux Lite 6.6 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.2
  * toSketch 1.0.1
  * POV-Ray-Rendering
  * fasteners 0.5.12
  * Manipulator 1.5.7
  * CADExchanger
  * A2plus 0.4.64a
  * ExplodedAssembly
  * boltsfc 2022.11.5
  * Launcher
  * Movie 2023.6.10
  * Render 2024.1.23
  * 3DfindIT 1.2.0
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
Not sure if this is related to FreeCAD itself, ideally you would report this in the Fasteners repository
Can you provide a demo file? Does it work in the stable FreeCAD version?
@shaise FYI

iulyus2004, 2024-05-19:
Hello!
Thank you very much for your reply.
Apologize for my late reply...just I tested the function on stable version and still not working. I will attach an example for you to analyze it. 
The only way to work around this problem, is to use arrays option from Draft module and to multiply the fasteners from the main part, which is ok, but not ideal :).

[Fasteners arrays issue.zip](https://github.com/FreeCAD/FreeCAD/files/15370928/Fasteners.arrays.issue.zip)

maxwxyz, 2024-05-20:
ah, now I understand your issue and can reproduce. So you want to move the fastener to the other locations which are a Draft Aray and they cannot be aligned there.
@shaise I guess this should be in the Fasteners repository, do you agree?

maxwxyz, 2024-05-20:
@Roy-043 maybe can you weigh in? I've found this old issue in the Fasteners repo which is closed. https://github.com/shaise/FreeCAD_FastenersWB/issues/50
Should it work with Draft Arrays? Because in the demo file above all fasteners are at the original location.

Roy-043, 2024-05-20:
I would create a Std_Part from the bracket and fasteners and array that. But that does not answer the problem.

The extra washers are not invisible, they are just not attached to the correct edge. This is not a Draft issue but a Fasteners issue.

iulyus2004, 2024-05-20:
Hi all! Thank you very much for your support.

I don't think that I was very explicit, so you can blame my English, or my poor skill on giving explanations. :). 
For this I created a video for you to see and analyze what is wrong. In my opinion I think that the compatibility between A2plus and Draft module is the one which is causing the problem. Unfortunately, I don't have the skills on using other assembly modules, as A2 plus is very appropriate to Solidworks software.
As I am a Solidworks user during the day at work, I will try to explain what I am trying to do, but using some Solidworks language.
 So, I created an assembly, and as an equivalent for Linear Pattern, I used Array option from Draft to replicate the function.
As you can see in the video, on the (I will call it "Parent"), I accessed the Fasteners and I added two washers on the parent. After, I tried to add in manual mode another two washers, in one of the arrayed brackets. As you can see on the tree, they seem to exist on the assembly but they are not visible. Not visible on the video, but the washers were linked to the parent and not to arrayed bracket.
After, I duplicate one of the brackets, and for exemplification, I was trying to use some mates to attach it to one of the arrayed brackets. As you can see, all the mates are going to be referenced to the Parent and not to arrayed bracket. This was causing me some issues in one of my complex assemblies in Freecad, as sometimes it depends of the assembly, you want to refer specific parts to the arrayed parts, and not to parent. The only way at that time was to reference everything to the parent and visually to correspond with the position of the arrayed item., which to be honest it is a bit annoying.

https://github.com/FreeCAD/FreeCAD/assets/170055309/239d767c-2ef6-45c2-a833-fa414f0917d3

Roy-043, 2024-05-21:
Thanks for the video. It confirms that I have understood the problem correctly. It is related to the use of Link arrays in Draft. The addons you use are not able to handle them properly. An easy solution is to uncheck the "Link array" checkbox when creating the array. This will create a less efficient 'regular' array. See https://wiki.freecad.org/Draft_OrthoArray.

In the attached zip there are two files for the Fastener WB that should fix the issue for that WB. They should be put in the `.../FreeCAD/Mod/fasteners/` folder.
[Fasteners_update.zip](https://github.com/FreeCAD/FreeCAD/files/15390633/Fasteners_update.zip)

Roy-043, 2024-05-21:
@kbwbe FYI

maxwxyz, 2024-05-21:
@Roy-043 can I close this one here as it will be fixed in the fasteners repository?

Roy-043, 2024-05-21:
Yes, this can be closed.

closed 2024-05-21 by Roy-043