FreeCAD/FreeCAD #14069: Python backtrace in findPlacement when solving integrated assembly with modified model

johnsonm, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to re-solve an Assembly after a change leads to a backtrace.

* Clone https://gitlab.com/mcdanlj/RotaryBroach
* Open RotaryBroachParts.FCStd and RotaryBroachAssembly.FCStd
* In RotaryBroachParts, select the **Broach** and change the Size parameter from ".5x1.75" to "8x28"

[image]

[image]

* In RotaryBroachAssembly, enter the Assembly workbench, double-click on the Assembly, and choose Assembly_SolveAssembly (Z)

```
Running the Python command 'Assembly_SolveAssembly' failed:
Traceback (most recent call last):
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/CommandSolveAssembly.py", line 71, in Activated
    assembly.solve()
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/JointObject.py", line 523, in updateJCSPlacements
    joint.Placement2 = self.findPlacement(
                       ^^^^^^^^^^^^^^^^^^^
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/JointObject.py", line 544, in findPlacement
    plc = UtilsAssembly.findPlacement(obj, part, elt, vtx, ignoreVertex)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/UtilsAssembly.py", line 891, in findPlacement
    edge = obj.Shape.Edges[elt_index - 1]
           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^

list index out of range
```

It's possible that the model changes break the constraints, and the problem is only a traceback where there should be an error, of course.

The original files were made a few months ago in various weekly builds of 0.22 dev.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git)
Build type: Unknown
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2

Built with FC_USE_TNP_FIX
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
@PaddleStroke FYI

johnsonm, 2024-06-03:
I have validated that this still occurs at 77fc1de6

PaddleStroke, 2024-06-04:
please assign me @maxwxyz

johnsonm, 2025-06-28:
I can't reproduce exactly because FreeCAD at 352fadd5c7 now complains about having written loads of corrupt 2D references and fails in the first step of changing the configuration. If I fix up the document and reproduce something like it, I can open a new, up-to-date issue later. I'll close this for now.

closed 2025-06-28 by johnsonm