FreeCAD/FreeCAD #14072: BIM: provide migration to new BIM workbench

furgo16, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After Arch becoming BIM, one of the effects is that the originally external BIM workbench is now part of the FreeCAD install.

After upgrading to a version of FreeCAD with the built-in BIM workbench, users who installed the external BIM workbench in the past, are welcomed with a Python traceback on the Report view and/or the notification area. This is because the two workbenches clash: uninstalling the external BIM workbench solves the issue.

A usability improvement would be to warn users if the external workbench is detected, and suggest to remove it.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git) Snap 1022
Build type: Release
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
  * OpenTheme.backup1715920329.3075469 2024.4.20 (Disabled)
```

### Subproject(s) affected?

Arch

### Anything else?

Traceback that pops up if the two BIM workbenches are present in the system:

```python-traceback
06:15:35  No module named 'Arch_rc'Error: Initializing one or more of the Draft modules failed, Draft will not work as expected.
06:15:35  No module named 'Arch_rc'
06:15:35  Traceback (most recent call last):
File "/snap/freecad/1022/usr/Mod/Draft/InitGui.py", line 94, in Initialize
self.drawing_commands = it.get_draft_drawing_commands()
File "/snap/freecad/1022/usr/Mod/Draft/draftutils/init_tools.py", line 42, in get_draft_drawing_commands
from draftguitools import gui_arcs
File "/snap/freecad/1022/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)
File "/snap/freecad/1022/usr/Mod/Draft/draftguitools/gui_arcs.py", line 37, in <module>
import Draft
File "/snap/freecad/1022/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)
File "/snap/freecad/1022/usr/Mod/Draft/Draft.py", line 53, in <module>
from draftutils.utils import ARROW_TYPES as arrowtypes
File "/snap/freecad/1022/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)
File "/snap/freecad/1022/usr/Mod/Draft/draftutils/utils.py", line 42, in <module>
from draftutils import params
File "/snap/freecad/1022/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)
File "/snap/freecad/1022/usr/Mod/Draft/draftutils/params.py", line 31, in <module>
import Arch_rc
File "/snap/freecad/1022/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
does this also happen with the latest BIM Addon version?
@yorikvanhavre FYI

furgo16, 2024-05-17:
@maxwxyz this happened with BIM workbench version `2021.12.0`. I do updates regularly, and it's the latest as far as I can tell. Although IIRC the BIM workbench version number has been static for a while.

maxwxyz, 2024-05-17:
as I recall, there was a recent update of the addon to get rid of that problem. But for users that did not upgrade, they will have this error.

yorikvanhavre, 2024-05-17:
Indeed this last commit https://github.com/yorikvanhavre/BIM_Workbench/commit/02b509d6a66d5ad62a8d5b903dc5219dd7c27ff0 does show a popup message telling the user to uninstall the BIM addon. Unfortunately I see no way for the new BIM WB to detect this, as it is not loaded when the addon is there.

Anything we can do is in the BIM addon, which indeed needs to be updated....

But I think

1) We need to put a big warning in the release notes
2) Put a big warning in the BIM addon's README
3) We can suppose, when users meet this problem, they will go to the addon manager and try to update...

yorikvanhavre, 2024-05-17:
Or have another WB do this? Detect the clashing BIM addon and remove it? Maybe the addon installer could do that? What do you think @chennes ?

chennes, 2024-05-17:
That would rely on the user running the Addon Manager -- I'll mark BIM as obsolete, so the AM should warn you that you have a deprecated addon installed, but FreeCAD could also be more proactive, and during the calls to Init.py we could add code that manually checks for a user-installed BIM and warns about it. We can remove that code in the next release or something like that.

yorikvanhavre, 2024-05-17:
The thing is, which Init.py... Since the "new" BIM is never run because the old one takes precedence. Maybe Tux could do that? It does a couple of fixes om other WBs already.

maxwxyz, 2024-05-17:
@yorikvanhavre 
Also Arch WB seems still present along the BIM WB in the WB selector (maybe a config / WB sorting remainder?):
[image]
Choosing Arch results in: 
[image]

It's also in the preferences:
[image]

Shall I create a new issue for this?

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CfdOF 1.25.13
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14

```

furgo16, 2024-05-17:
It might be worth a new issue to investigate it properly. I cannot reproduce it on:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git) Snap 1022
Build type: Release
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
```

yorikvanhavre, 2024-05-17:
@maxwxyz you are compiling FreeCAD yourself, right? The Arch WB is then a leftover from a previous compilation (same happened when Path was renamed to CAM). Just remove Mod/Arch from your install or build folder.

This problem will not happen with users using a prebuilt package, so I don't think we need any special action about that...

maxwxyz, 2024-05-17:
@yorikvanhavre yes, thanks for clarification!

yorikvanhavre, 2024-05-27:
A new scenario is popping up that I had not foreseen... People don't want to delete the BIM addon because they also use a stable 0.21 version side-to-side... Of course, this should not be a problem anymore after publishing 1.0, but it's still annoying, and I'm not sure how best to solve it. Maybe the BIM addon should be renamed to something like BIM021? The problem is, how will that work for people who ALREADY have the BIM add-on installed...

paullee0, 2024-05-27:
@yorikvanhavre   Maybe 'update' the existing 'external BIM' so it detects whether it is running on 0.22/1.0 so it 'transfers back' to the built-in BIM, or even remove / rename 'itself'?

PR #14391 (2024-06-03): Tux: Detect and fix BIM addon
This PR uses the Tux addon (which in any case is already used to tweak stuff in Draft/Arch/BIM...) to detect a situation where both the BIM addon and the built-in BIM workbench are present.

If such a situation is met, the BIM addon path is renamed from UserAppDataDir/Mod/BIM to UserAppDataDir/Mod/BIM021 and the user gets a warning to restart FreeCAD.

Fixes #14072
commit c5a350ce (2024-06-03): Tux: Detect and fix BIM addon (#14391)

* Tux: Detect and fix BIM addon - fixes #14072

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2024-06-03 by chennes