FreeCAD/FreeCAD #14076: Segmentation fault using the Measure command

furgo16, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Open FreeCAD
2. Create a new document
3. Switch to Part workbench
4. Create a cube
5. Select the cube
6. Click on the Measure command (Std_Measure)
7. FreeCAD crashes (see stacktrace)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git) Snap 1022
Build type: Release
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

Core

### Anything else?

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f268f0b3520]
#1  0x7f26921d28cb in Gui::DAG::Model::slotChangeObject(Gui::ViewProviderDocumentObject const&, App::Property const&) from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x13b
#2  0x7f2691edc0a9 in boost::signals2::detail::signal_impl<void (Gui::ViewProviderDocumentObject const&, App::Property const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::ViewProviderDocumentObject const&, App::Property const&)>, boost::function<void (boost::signals2::connection const&, Gui::ViewProviderDocumentObject const&, App::Property const&)>, boost::signals2::mutex>::operator()(Gui::ViewProviderDocumentObject const&, App::Property const&) from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x1f9
#3  0x7f2691ebfa70 in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x200
#4  0x7f269147f6e9 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&, App::Property const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&, App::Property const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&, App::Property const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&, App::Property const&) from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x1f9
#5  0x7f26914a60ca in App::DocumentObject::onChanged(App::Property const*) from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x12a
#6  0x7f26915e8425 in App::Property::hasSetValue() from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x145
#7  0x7f269166853d in App::PropertyString::setValue(char const*) from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x12d
#8  0x7f2650f6c127 in MeasureGui::ViewProviderMeasureBase::updateData(App::Property const*) from /snap/freecad/1022/usr/lib/MeasureGui.so+0x4b7
#9  0x7f26923d4af0 in Gui::ViewProviderDocumentObject::updateView() from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x100
#10  0x7f2650f6c6f7 in MeasureGui::ViewProviderMeasure::positionAnno(Measure::MeasureBase const*) from /snap/freecad/1022/usr/lib/MeasureGui.so+0x127
#11  0x7f2691ec03d4 in Gui::Document::slotNewObject(App::DocumentObject const&) from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x504
#12  0x7f26914825e1 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&) from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x1d1
#13  0x7f26914402a9 in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /snap/freecad/1022/usr/lib/libFreeCADApp.so+0x299
#14  0x7f26920ac5aa in Gui::TaskMeasure::update() from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x8fa
#15  0x7f268f80c39e in QObject::event(QEvent*) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2ae
#16  0x7f2690378763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#17  0x7f2691f23798 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x98
#18  0x7f268f7dee4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#19  0x7f268f7e1f47 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x187
#20  /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x316ce7) [0x7f268f838ce7]
#21  /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f268d27bd3b]
#22  /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7f268d2d1258]
#23  /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f268d2793e3]
#24  0x7f268f838338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#25  0x7f268f7dd76b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#26  0x7f268f7e5d14 in QCoreApplication::exec() from /snap/freecad/1022/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#27  0x7f2691e5e2d4 in Gui::Application::runApplication() from /snap/freecad/1022/usr/lib/libFreeCADGui.so+0x6c4
#28  /snap/freecad/1022/usr/bin/FreeCAD(+0xa709) [0x55b3c674d709]
#29  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f268f09ad90]
#30  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f268f09ae40]
#31  /snap/freecad/1022/usr/bin/FreeCAD(+0xaac5) [0x55b3c674dac5]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
I cannot reproduce on Windows

xtemp09, 2024-05-17:
I cannot reproduce.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Debug
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

xtemp09, 2024-05-18:
Staring at the backtrace, I noticed `Gui::DAG::Model::slotChangeObject`. I activated _DAGView_ and managed to reproduce the crash. This issue is connected to #12163.

Here is the backtrace for Linux.

<details><summary>Backtrace</summary>

```cpp
Process 2557 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
    frame #0: 0x00007ffff32969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff32969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
    frame #1: 0x00007ffff32969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737256995456) at pthread_kill.c:78:10
    frame #2: 0x00007ffff32969b0 libc.so.6`__GI___pthread_kill(threadid=140737256995456, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff3242476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff32287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff322871b libc.so.6`__assert_fail_base.cold at assert.c:92:3
    frame #6: 0x00007ffff3239e96 libc.so.6`__GI___assert_fail(assertion="it != list.end()", file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/DAGView/DAGModelGraph.cpp", line=81, function="const Gui::DAG::GraphLinkRecord& Gui::DAG::findRecord(const Gui::ViewProviderDocumentObject*, const GraphLinkContainer&)") at assert.c:101:3
    frame #7: 0x00007ffff6f1ff30 libFreeCADGui.so`Gui::DAG::findRecord(VPDObjectIn=<unavailable>, containerIn=0x00005555599f6b30) at DAGModelGraph.cpp:81:3
    frame #8: 0x00007ffff6f1365a libFreeCADGui.so`Gui::DAG::Model::slotChangeObject(this=0x000055555833ad80, VPDObjectIn=0x000055555a6ef2b0, propertyIn=0x000055555a6eed88) at DAGModel.cpp:311:47
    frame #9: 0x00007ffff6f1c268 libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::DAG::Model::*&)(Gui::ViewProviderDocumentObject const&, App::Property const&), Gui::DAG::Model*&, Gui::ViewProviderDocumentObject const&, App::Property const&>((null)=__invoke_memfun_deref @ 0x00007fffffffad50, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #10: 0x00007ffff6f1c290 libFreeCADGui.so`void std::_Bind<void (Gui::DAG::Model::* (Gui::DAG::Model*, std::_Placeholder<1>, std::_Placeholder<2>))(Gui::ViewProviderDocumentObject const&, App::Property const&)>::operator()<Gui::ViewProviderDocumentObject const&, App::Property const&, void>(Gui::ViewProviderDocumentObject const&, App::Property const&) [inlined] std::__invoke_result<void (Gui::DAG::Model::*&)(Gui::ViewProviderDocumentObject const&, App::Property const&), Gui::DAG::Model*&, Gui::ViewProviderDocumentObject const&, App::Property const&>::type std::__invoke<void (Gui::DAG::Model::*&)(Gui::ViewProviderDocumentObject const&, App::Property const&), Gui::DAG::Model*&, Gui::ViewProviderDocumentObject const&, App::Property const&>((null)=0x000055555a6eed88, (null)=0x000055555a6ef2b0, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #11: 0x00007ffff6f1c285 libFreeCADGui.so`void std::_Bind<void (Gui::DAG::Model::* (Gui::DAG::Model*, std::_Placeholder<1>, std::_Placeholder<2>))(Gui::ViewProviderDocumentObject const&, App::Property const&)>::operator()<Gui::ViewProviderDocumentObject const&, App::Property const&, void>(Gui::ViewProviderDocumentObject const&, App::Property const&) [inlined] void std::_Bind<void (Gui::DAG::Model::* (Gui::DAG::Model*, std::_Placeholder<1>, std::_Placeholder<2>))(Gui::ViewProviderDocumentObject const&, App::Property const&)>::__call<void, Gui::ViewProviderDocumentObject const&, App::Property const&, 0ul, 1ul, 2ul>((null)=<unavailable>, __args=0x00007fffffffad50, this=<unavailable>) at functional:506:24
    frame #12: 0x00007ffff6f1c281 libFreeCADGui.so`void std::_Bind<void (Gui::DAG::Model::* (Gui::DAG::Model*, std::_Placeholder<1>, std::_Placeholder<2>))(Gui::ViewProviderDocumentObject const&, App::Property const&)>::operator()<Gui::ViewProviderDocumentObject const&, App::Property const&, void>(this=<unavailable>, (null)=0x000055555a6ef2b0, (null)=0x000055555a6eed88) at functional:591:32
    frame #13: 0x00007ffff6f1c29f libFreeCADGui.so`boost::detail::function::void_function_obj_invoker2<std::_Bind<void (Gui::DAG::Model::* (Gui::DAG::Model*, std::_Placeholder<1>, std::_Placeholder<2>))(Gui::ViewProviderDocumentObject const&, App::Property const&)>, void, Gui::ViewProviderDocumentObject const&, App::Property const&>::invoke(function_obj_ptr=<unavailable>, a0=<unavailable>, a1=<unavailable>) at function_template.hpp:158:11
    frame #14: 0x00007ffff6cdb8a7 libFreeCADGui.so`boost::function2<void, Gui::ViewProviderDocumentObject const&, App::Property const&>::operator()(this=<unavailable>, a0=<unavailable>, a1=<unavailable>) const at function_template.hpp:763:28
    frame #15: 0x00007ffff6cdb969 libFreeCADGui.so`boost::signals2::detail::void_type boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::ViewProviderDocumentObject const&, App::Property const&>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::ViewProviderDocumentObject const&, App::Property const&), boost::function<void (Gui::ViewProviderDocumentObject const&, App::Property const&)>>, boost::signals2::mutex>>>(boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost:: [... cut ...]
    frame #101: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #102: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)

```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Debug
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

One may notice that it is failing assert causes the segfault.

furgo16, 2024-05-19:
At this point, I cannot use the Measure tool. It crashes FreeCAD every time I click on it. Let me know if I can provide more info or help testing.

furgo16, 2024-05-19:
As a workaround, _disabling_ DAGView as per https://wiki.freecad.org/DAG_view stops the Measure tool from crashing. Thanks xtemp09 for the hint, I can now continue working on my model.

maxwxyz, 2024-05-19:
@hlorus can you reproduce this?

hlorus, 2024-05-20:
> @hlorus can you reproduce this?

Yes i can reproduce it when enabling DAG View, i'll have to investigate.

krushia, 2024-05-23:
Introduced sometime between these builds:

(Last weekly without crash)
Version: 0.22.0dev.37249 (Git) AppImage
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6 

(First weekly with crash)
Version: 0.22.0dev.37302 (Git) AppImage
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977

hlorus, 2024-05-25:
Thanks @krushia! It was introduced in 2b9e9d6ca50e54486953706c4cd3c6cc08b8a22e, more specifically line 310 seems to trigger the crash.

luzpaz, 2024-07-24:
Path forward ?

hlorus, 2024-07-24:
This seems to be an issue in DAG View which i'm not familiar with, maybe there's someone who knows that area better?

maxwxyz, 2024-07-24:
would you consider this a blocker for a 1.0 release?

luzpaz, 2024-09-24:
This could be a blocker since folks will be exploring the rewritten measurement functionality and this crashes FC.

bgbsww, 2024-09-26:
I'm no QT expert, but https://github.com/bgbsww/FreeCAD/blob/69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa/src/Mod/Measure/App/MeasurePosition.cpp#L126-L149 looks funny to me.  Is it safe to just return the local variable text, or do we need to do something to avoid returning memory that can be overwritten?

hlorus, 2024-09-26:
I don't see an issue, as long as we pass by value the object is just copied.

bgbsww, 2024-09-26:
Fair enough.  I can't replicate this, but I would try checking that the obj->getResultString() value at https://github.com/bgbsww/FreeCAD/blob/69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa/src/Mod/Measure/Gui/ViewProviderMeasureBase.cpp#L369 is legitimate.

That, plus some sporadic online notes that toStdString() can segfault on invalid QStrings led me to that question.  You can probably at least defend the segfault and move on.  Does something about the QTextStream which apparently uses a fixed internal buffer lead to an invalid QString in strange cases?

On reading Obj can't be bad; name can't be bad, we aren't getting into setValue(), so that leaves me one choice.

hlorus, 2024-09-30:
> Fair enough.  I can't replicate this, but I would try checking that the obj->getResultString() value at https://github.com/bgbsww/FreeCAD/blob/69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa/src/Mod/Measure/Gui/ViewProviderMeasureBase.cpp#L369 is legitimate.
> 
> That, plus some sporadic online notes that toStdString() can segfault on invalid QStrings led me to that question.  You can probably at least defend the segfault and move on.  Does something about the QTextStream which apparently uses a fixed internal buffer lead to an invalid QString in strange cases?
> 
> On reading Obj can't be bad; name can't be bad, we aren't getting into setValue(), so that leaves me one choice.
> 
> ETA:  Also, this is the only place in the code base I see that particular combination of QString and QTextStream used to return a method value.  The other thought is that a .flush() might help.

Did you enconter a crash/erroe which is provoked from the method you mentioned?
It might be better to open a new issue for what you've mentioned. This issue is about the crash which happens in combination with the DAG View.

hyarion, 2024-10-03:
I just tried it on main branch head.

I get an error, but not a crash:

```
20:12:59  pyException: Traceback (most recent call last):
  File "/Users/nauck/projects/freecad/build-debug/Mod/Measure/MeasureCOM.py", line 114, in parseSelection
    obj.Element = (o, item["subName"])
    ^^^^^^^^^^^
<class 'ValueError'>: PropertyLinkSub does not support external object
```

Session screen capture:

https://github.com/user-attachments/assets/6d7bc397-4360-4ea1-9f76-98e9e6820045

Full info:
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 (Git)
Build type: Release
Branch: main
Hash: d423795e2dc32dc16f1e8c1835104deb01c9cab9
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

zbynekwinkler, 2024-10-05:
I am seeing the same behavior as @hyarion (on Debian + conda, no crash, "just" exception).

furgo16, 2024-10-05:
I can confirm that with a recent git version, the segmentation fault does no longer occur, but a Python exception instead:

```python-traceback
pyException: Traceback (most recent call last):
  File "/snap/freecad/1156/usr/Mod/Measure/MeasureCOM.py", line 114, in parseSelection
    obj.Element = (o, item["subName"])
<class 'ValueError'>: PropertyLinkSub does not support external object
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1156
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

hlorus, 2024-10-05:
That exception is likley just a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16699

maxwxyz, 2024-10-06:
I'll close this then.

closed 2024-10-06 by maxwxyz