FreeCAD/FreeCAD #14084: BIM workbench is not fully documented

furgo16, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Now that the Arch workbench is becoming the BIM workbench, there are a number of additional commands in the later that need to be documented.

This issue collects an overview and a list of tasks to organize the documentation effort.

## Documentation tasks

- [ ] Example documentation issue

## Overview: documentation status

| Toolbar | Command | Documented | Notes |
| --- | --- | --- | --- |
| Drafting tools | [BIM_Sketch](https://wiki.freecad.org/BIM_Sketch) | :x: |   |
| Drafting tools | [Draft_Line](https://wiki.freecad.org/Draft_Line) | :white_check_mark: |   |
| Drafting tools | [Draft_Wire](https://wiki.freecad.org/Draft_Wire) | :white_check_mark: |   |
| Drafting tools | [Draft_Circle](https://wiki.freecad.org/Draft_Circle) | :white_check_mark: |   |
| Drafting tools | [Draft_Arc](https://wiki.freecad.org/Draft_Arc) | :white_check_mark: |   |
| Drafting tools | [Draft_Arc_3Points](https://wiki.freecad.org/Draft_Arc_3Points) | :white_check_mark: |   |
| Drafting tools | [Draft_Fillet](https://wiki.freecad.org/Draft_Fillet) | :white_check_mark: |   |
| Drafting tools | [Draft_Ellipse](https://wiki.freecad.org/Draft_Ellipse) | :white_check_mark: |   |
| Drafting tools | [Draft_Polygon](https://wiki.freecad.org/Draft_Polygon) | :white_check_mark: |   |
| Drafting tools | [Draft_Rectangle](https://wiki.freecad.org/Draft_Rectangle) | :white_check_mark: |   |
| Drafting tools | [Draft_BSpline](https://wiki.freecad.org/Draft_BSpline) | :white_check_mark: |   |
| Drafting tools | [Draft_BezCurve](https://wiki.freecad.org/Draft_BezCurve) | :white_check_mark: |   |
| Drafting tools | [Draft_CubicBezCurve](https://wiki.freecad.org/Draft_CubicBezCurve) | :white_check_mark: |   |
| Drafting tools | [Draft_Point](https://wiki.freecad.org/Draft_Point) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Lock](https://wiki.freecad.org/Draft_Snap_Lock) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Endpoint](https://wiki.freecad.org/Draft_Snap_Endpoint) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Midpoint](https://wiki.freecad.org/Draft_Snap_Midpoint) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Center](https://wiki.freecad.org/Draft_Snap_Center) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Angle](https://wiki.freecad.org/Draft_Snap_Angle) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Intersection](https://wiki.freecad.org/Draft_Snap_Intersection) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Perpendicular](https://wiki.freecad.org/Draft_Snap_Perpendicular) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Extension](https://wiki.freecad.org/Draft_Snap_Extension) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Parallel](https://wiki.freecad.org/Draft_Snap_Parallel) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Special](https://wiki.freecad.org/Draft_Snap_Special) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Near](https://wiki.freecad.org/Draft_Snap_Near) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Ortho](https://wiki.freecad.org/Draft_Snap_Ortho) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Grid](https://wiki.freecad.org/Draft_Snap_Grid) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_WorkingPlane](https://wiki.freecad.org/Draft_Snap_WorkingPlane) | :white_check_mark: |   |
| Draft snap | [Draft_Snap_Dimensions](https://wiki.freecad.org/Draft_Snap_Dimensions) | :white_check_mark: |   |
| Draft snap | [Draft_ToggleGrid](https://wiki.freecad.org/Draft_ToggleGrid) | :white_check_mark: |   |
| 3D/BIM tools | [BIM_Project](https://wiki.freecad.org/BIM_Project) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Site](https://wiki.freecad.org/Arch_Site) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Building](https://wiki.freecad.org/Arch_Building) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Level](https://wiki.freecad.org/Arch_Level) | :x: |   |
| 3D/BIM tools | [Arch_Space](https://wiki.freecad.org/Arch_Space) | :white_check_mark: |   |
| 3D/BIM tools | [Separator](https://wiki.freecad.org/Separator) | :x: |   |
| 3D/BIM tools | [Arch_Wall](https://wiki.freecad.org/Arch_Wall) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_CurtainWall](https://wiki.freecad.org/Arch_CurtainWall) | :white_check_mark: |   |
| 3D/BIM tools | [BIM_Column](https://wiki.freecad.org/BIM_Column) | :x: |   |
| 3D/BIM tools | [BIM_Beam](https://wiki.freecad.org/BIM_Beam) | :x: |   |
| 3D/BIM tools | [BIM_Slab](https://wiki.freecad.org/BIM_Slab) | :x: |   |
| 3D/BIM tools | [BIM_Door](https://wiki.freecad.org/BIM_Door) | :x: |   |
| 3D/BIM tools | [Arch_Window](https://wiki.freecad.org/Arch_Window) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Pipe](https://wiki.freecad.org/Arch_Pipe) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_PipeConnector](https://wiki.freecad.org/Arch_PipeConnector) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Stairs](https://wiki.freecad.org/Arch_Stairs) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Roof](https://wiki.freecad.org/Arch_Roof) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Panel](https://wiki.freecad.org/Arch_Panel) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Frame](https://wiki.freecad.org/Arch_Frame) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Fence](https://wiki.freecad.org/Arch_Fence) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Truss](https://wiki.freecad.org/Arch_Truss) | :white_check_mark: |   |
| 3D/BIM tools | [Arch_Rebar](https://wiki.freecad.org/Arch_Rebar) | :white_check_mark: |   |
[... 91 lines cut ...]
freecad_docs_url = "https://wiki.freecad.org"
header = "| Toolbar | Command | Documented | Notes |\n| --- | --- | --- | --- |"
header_printed = False
workbench = Gui.activeWorkbench()
toolbar_items = workbench.getToolbarItems()
for toolbar in toolbar_items:
    for command in toolbar_items[toolbar]:
        if toolbar not in common_toolbars:
            if not header_printed:
                print(header)
                header_printed = True
            response = requests.head(f"{freecad_docs_url}/{command}")
            if response.status_code == 200:
                is_documented = True
            elif 400 <= response.status_code < 500:
                is_documented = False
            else:
                is_documented = None
            is_documented = ':white_check_mark:' if is_documented else ':x:'
            print("|", toolbar, "|", f"[{command}]({freecad_docs_url}/{command})", "|", is_documented, "|", " ", "|")
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-05-17:
It is new to me that we use GitHub to track the documentation effort. When was that decided??

Roy-043, 2024-05-17:
Please stop creating related issues.

yorikvanhavre, 2024-05-17:
> It is new to me that we use GitHub to track the documentation effort. When was that decided??

I don't think it was, but why not? I don't see well where else it could be tracked efficiently, and it might be a good thing so we try to have the doc in shape for 1.0.

The day the doc is in its own repo, of course, that will be better, that's where we'll track things...

maxwxyz, 2024-05-17:
@furgo16 could you please use this issue for tracking documentation? I've noticed you started creating issues for every command.

Roy-043, 2024-05-17:
> but why not?

I can think of several reasons. But we should not have that discussion here.

yorikvanhavre, 2024-05-17:
> I can think of several reasons. But we should not have that discussion here.

Agreed. Let's bring the topic up during next M3 meeting maybe

yorikvanhavre, 2024-05-17:
@furgo16 nice script BTW! Maybe something like that could also be used to regenerate the tools table at https://wiki.freecad.org/BIM_Workbench

furgo16, 2024-05-17:
I filed this issue with the spirit of contributing. The intention was to serve as the central point to organize the effort (similar to a GitLab epic) , with an [issue task list](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/about-task-lists#about-issue-task-lists) and issues for every command, to more easily track the status. But also importantly to encourage other contributors to pick a task and complete the BIM documentation collaboratively. Separate issues could also be used in [projects](https://github.com/FreeCAD/FreeCAD/projects), similarly to what was done with the TNP effort.

Regardless of the platform, I personally can only see advantages in using the same tools to manage documentation as code, especially if there is eventually a migration to markdown files for docs. Yet others might see this differently, which is absolutely fine. Constructive debate is one of the great things about open source.

I was not anticipating filing this issue could stir negative feelings. Apologies for that, as it was not my intention. Feel free to close this issue if not desired. No hard feelings, happy to keep contributing in another area of the project if that's useful. Thanks.

furgo16, 2024-05-17:
> @furgo16 nice script BTW! Maybe something like that could also be used to regenerate the tools table at https://wiki.freecad.org/BIM_Workbench

Thanks, I had thought about that as well! I would have liked to add each command's icon to the table. In the end, I could not figure out a way to retrieve the icons (and to upload them without manual steps), so I kept it simple as it is.

krushia, 2024-05-19:
A few notes from a clueless BIM user

1. I got a good laugh when seeing the "first run" dialog for BIM today, which has a link to "complete documentation" (sorry didn't get a screenshot). Clicking the link opens https://wiki.freecad.org/BIM_Workbench where the first words read "This documentation is not finished." I don't have the heart to open a bug just for that nitpick :)
2. Whatever happened to the GSoC documentation overhaul project from last year? From what I've read, it was supposed to be the path forward.

yorikvanhavre, 2024-05-21:
Documentation has a lot of those quirks :sweat_smile: Hopefully very soon we'll remove that "documentation is in process" note :) The welcome screen also needs a rework anyway.

Regarding last year's GSoC project, it basically still lacks functionality and didn't reach yet all the goals we wanted, so it's kind of stalled currently, and we are still using the wiki for now. But indeed after the release we need to retake that effort

Roy-043, 2024-06-13:
FYI:
The BIM Wiki documentation has been provisionally updated. See [BIM Workbench](https://wiki.freecad.org/BIM_Workbench). Missing pages appear in red. Note that 7 tools in the Modify menu are wrappers for Part tools. Perhaps they can be removed and the Part tools used directly (same as has been done with [Part Builder](https://wiki.freecad.org/Part_Builder))

yorikvanhavre, 2024-06-24:
Excellent! I'll give it a push this week too.

Regarding the "wrapped" Part tools, the original idea was to be able to late-load the Part WB so BIM starts fast. Not sure how relevant that is...

luzpaz, 2024-08-30:
@yorikvanhavre any thoughts on having a documentation hackathon during the 1.0RC process ?

yorikvanhavre, 2024-09-02:
That's a good idea. I would say it's in pretty good shape already, thanks mostly to @Roy-043 , but indeed we could use the next two weeks to do a bit of a sprint.

I'm gathering other things that should be announced/tweeted/shout for these next two weeks, I'll pass that to everybody