FreeCAD/FreeCAD #14090: BIM: DialogIfcProperties.ui occurs twice in src (with and without cap)

Roy-043, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are 2 DialogIfcProperties.ui files (with and without cap).  That is problematic for developers using the Windows OS. And this is probably not intended anyway.

@yorikvanhavre FYI.

### Full version info

```shell
V1.0
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

superfloh247, 2024-05-17:
MacOS also affected

  'src/Mod/BIM/Resources/ui/DialogIfcProperties.ui'
  'src/Mod/BIM/Resources/ui/dialogIfcProperties.ui'

Roy-043, 2024-05-17:
On Windows the problem is more severe. The file system is insensitive on Windows. So there can only be one DialogIfcProperties.ui file. And GitKraken (the git GUI application I use) therefore thinks that I have deleted a file.

maxwxyz, 2024-05-17:
Can confirm, this is problematic on Windows

maxwxyz, 2024-05-17:
any idea how I can solve this on windows? I cannot rebase or push as git complains with every action. When I revert changes it pops back automatically.

yorikvanhavre, 2024-05-17:
I don't understand what you are trying to do? (BTW I made a PR that solves the issue - #14091 )

maxwxyz, 2024-05-17:
The file always pops up and says it has unsaved changes so I should stage them to commit or revert changes. When I revert them, the changes come back immediately.
Also when trying to rebase a branch on main same problem:
` error: cannot pull with rebase: You have unstaged changes.`

Roy-043, 2024-05-17:
If you reset main the problem should go away. You have to reset 'hard'.

closed 2024-05-17 by Roy-043

maxwxyz, 2024-05-17:
Unfortunately not, any other ideas?

Roy-043, 2024-05-17:
Sorry, I am not a GitHub specialist (or fan for that matter). With GitKraken 'hard' resetting fixes the issue.

maxwxyz, 2024-05-17:
@Roy-043 ok it worked now I had to reset before the merge of the big PR

PR #14091 (2024-05-17): BIM: Renamed duplicate DialogIfcProperties
Two dialogs had the same file name with different cases
Fixes #14090
commit c520c492 (2024-05-17): BIM: Renamed duplicate DialogIfcProperties - fixes #14090 (#14091)