FreeCAD/FreeCAD #14101: Gui: Crash on Python Console autocompletion Qt6/Pyside6

pieterhijma, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The autocompletion popup in the Python console causes a segfault with PySide6/Qt6.

To reproduce, type the following in the Python console:

```python
from PySide import QtNetwork
QtNetwork.QNetworkRequest.
```
On typing the `.`, the console tries to show a popup with autocompletion, but it crashes before it can do that:

Stacktrace:

```
#0  /usr/lib/libc.so.6(+0x3cae0) [0x7a1d61250ae0]
#1  /usr/lib/libpython3.12.so.1.0(PyFrame_GetCode+0x4) [0x7a1d63e6ec14]
#2  /usr/lib/libshiboken6.cpython-312-x86_64-linux-gnu.so.6.7(+0x32656) [0x7a1d63bc0656]
#3  /usr/lib/libshiboken6.cpython-312-x86_64-linux-gnu.so.6.7(mangled_type_getattro+0x135) [0x7a1d63bc0a15]
#4  /usr/lib/libpython3.12.so.1.0(PyObject_GetAttrString+0xb3) [0x7a1d63dad623]
#5  0x7a1d6757747a in Py::Object::getAttr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x36
#6  0x7a1d67b125c5 in Gui::CallTipsList::extractTipsFromObject(Py::Object&, Py::List&, QMap<QString, Gui::CallTip>&) const from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0xf3
#7  0x7a1d67b12159 in Gui::CallTipsList::extractTips(QString const&) const from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0xb4f
#8  0x7a1d67b13f2a in Gui::CallTipsList::showTips(QString const&) from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x70e
#9  0x7a1d67b26c79 in Gui::PythonConsole::keyPressEvent(QKeyEvent*) from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x62f
#10  0x7a1d6294950e in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x62e
#11  0x7a1d629dbf6a in QFrame::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x3a
#12  0x7a1d628fc44d in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x4d
#13  0x7a1d6290187b in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x124b
#14  0x7a1d677196c4 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x114
#15  0x7a1d6193fe18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#16  /usr/lib/libQt6Widgets.so.6(+0x160336) [0x7a1d62960336]
#17  0x7a1d628fc44d in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x4d
#18  0x7a1d677196c4 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x114
#19  0x7a1d6193fe18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#20  0x7a1d61f77a30 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/libQt6Gui.so.6+0x330
#21  0x7a1d61fe8dcc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x2ac
#22  /usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x4dcb7) [0x7a1d5d07ccb7]
#23  /usr/lib/libglib-2.0.so.0(+0x5ca89) [0x7a1d5e70ea89]
#24  /usr/lib/libglib-2.0.so.0(+0xbe9b7) [0x7a1d5e7709b7]
#25  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7a1d5e70df95]
#26  0x7a1d61b93389 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x89
#27  0x7a1d61948350 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1b0
#28  0x7a1d61943c1d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#29  /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so(+0x1372470) [0x7a1d67572470]
#30  /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so(+0x13727bb) [0x7a1d675727bb]
#31  0x7a1d67572b74 in Gui::Application::runApplication() from /home/pieter/build-freecad-main-debug/lib/libFreeCADGui.so+0x22e
#32  ./bin/FreeCAD(+0x26579) [0x607c1498b579]
#33  /usr/lib/libc.so.6(+0x25c88) [0x7a1d61239c88]
#34  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7a1d61239d4c]
#35  ./bin/FreeCAD(+0x25745) [0x607c1498a745]
```

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37347 (Git)
Build type: DEBUG
Branch: main
Hash: 18c85a58aeb0fa9d69fae1b70de85d7fff9e0eb3
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

This seems related: #10974

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-05-17:
@chennes, we briefly discussed this and mentioned you had a Qt debug compilation, so let's hope you can reproduce this.

maxwxyz, 2024-05-17:
related: https://github.com/FreeCAD/FreeCAD/issues/13303 ?

pieterhijma, 2024-05-17:
Yes, I think so.

Jookia, 2024-06-21:
Taking down some notes here: It seems PyFrame_GetCode is being called with a NULL pointer passed from PyEval_GetFrame.

```
(gdb) disas

Dump of assembler code for function PyFrame_GetCode:
   0x00007ffff4c6ec10 <+0>:     endbr64
=> 0x00007ffff4c6ec14 <+4>:     mov    rax,QWORD PTR [rdi+0x18]
   0x00007ffff4c6ec18 <+8>:     mov    rdx,QWORD PTR [rax]
   0x00007ffff4c6ec1b <+11>:    mov    eax,DWORD PTR [rdx]
   0x00007ffff4c6ec1d <+13>:    add    eax,0x1
   0x00007ffff4c6ec20 <+16>:    je     0x7ffff4c6ec24 <PyFrame_GetCode+20>
   0x00007ffff4c6ec22 <+18>:    mov    DWORD PTR [rdx],eax
   0x00007ffff4c6ec24 <+20>:    mov    rax,rdx
   0x00007ffff4c6ec27 <+23>:    ret
End of assembler dump.
(gdb) p $rdi
$49 = 0

(gdb) x/40i $pc-64

   0x7ffff591ef58:      iret
   0x7ffff591ef59:      add    eax,DWORD PTR [rax]
   0x7ffff591ef5b:      test   eax,eax
   0x7ffff591ef5d:      jne    0x7ffff591f0c0
   0x7ffff591ef63:      movzx  eax,BYTE PTR [rip+0x3e12e]        # 0x7ffff595d098
   0x7ffff591ef6a:      test   al,al
   0x7ffff591ef6c:      jne    0x7ffff591ef86
   0x7ffff591ef6e:      lea    rbx,[rip+0x3e123]        # 0x7ffff595d098 # mutex of some type? value 1
   0x7ffff591ef75:      mov    rdi,rbx
   0x7ffff591ef78:      call   QWORD PTR [rip+0x3ceea]        # 0x7ffff595be68 # <__cxxabiv1::__cxa_guard_acquire(__cxxabiv1::__guard*)>
   0x7ffff591ef7e:      test   eax,eax
   0x7ffff591ef80:      jne    0x7ffff591f0e8
   0x7ffff591ef86:      call   QWORD PTR [rip+0x3c4c4]        # 0x7ffff595b450 # PyEval_GetFrame
   0x7ffff591ef8c:      mov    rbx,rax
   0x7ffff591ef8f:      mov    rdi,rax
   0x7ffff591ef92:      call   QWORD PTR [rip+0x3cf90]        # 0x7ffff595bf28 # PyFrame_GetCode
=> 0x7ffff591ef98:      mov    rdi,rax
   0x7ffff591ef9b:      call   QWORD PTR [rip+0x3c4ef]        # 0x7ffff595b490
   0x7ffff591efa1:      mov    rdi,rbx

```

This looks close enough to https://github.com/pyside/pyside-setup/blob/dev/sources/shiboken6/libshiboken/sbkfeature_base.cpp#L124 and we can see it doesn't check for NULL. It seems to be a regression introduced in 2022, shipping in 6.4.0: https://github.com/pyside/pyside-setup/commit/615d6a820137b31f03295e884f58cd46aadf5032

I've done the work of opening a bug upstream: https://bugreports.qt.io/browse/PYSIDE-2796

Jookia, 2024-06-22:
I've made a workaround in #14864. Please test it.

pieterhijma, 2024-06-24:
@Jookia, thank you for your analysis and PR! I'm looking at it.

commit 8e147669 (2024-07-15): GUI: Work around Shiboken 6 bug for CallTips

Shiboken 6.4.0 to 6.8.0 and newer crash when using getAttr on their
objects. On these versions and newer, use a wrapper function to
call getattr.

Fixes #14101
closed 2024-07-15 by chennes