FreeCAD/FreeCAD #14102: Assembly: TNP

PaddleStroke, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The assemblies can break easily because of TNP.

It is because the joints are computing the placements based on element names stored as string. For example a joint will store : 
Object1 = "Body1"
Element1 = "Face10"
Vertex1 = "Vertex8"

Object1 = "Body2"
Element1 = "Edge5"
Vertex1 = "Vertex8"

So my question is how to solve TNP here :) @CalligaroV @bgbsww if you can give me some hints perhaps?
Thanks

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-05-17:
I'll try to stay available on Discord ( bgbsww ) so we can discuss in real time.

I don't know much about Assembly.  Are we talking across files or within a file?

In any case, the elementMaps in the TNP code associate a history with the current name of an element.  The history is the sequence of operations that caused a shape to exist, like making a box, making a cylinder, and doing a boolean between them is conceptually a three step history that describes a shape.  There are both forward ( history to current name ) and reverse element maps (current name to history ), but most importantly, there's a lookup and replacement of references when something changes.  So if I've built something else on top of that shape above, and then I change that shape, I call back and change all the references to that shape to point at the new name.

That's probably what you want to tap into here - getting your references changed when an underlying shape changes.

PaddleStroke, 2024-05-27:
@bgbsww I made a small video to explain the TNP issue in assembly : 

https://github.com/FreeCAD/FreeCAD/assets/19984177/31267008-f965-41e2-8ad7-504f8ad5d4bb

If you need more details, let me know when you're available so we can hope in a video chat.
Thanks

bgbsww, 2024-05-31:
Thank you for the video, it helps me understand a lot.  I suspect the root of this issue is around the Element1,2 and Vertex1,2 properties which appear to be PropertyStrings and not PropertyLinks.  There is a lot of code around PropertyLinkBase::_updateElementReference that essentially rewrites the links to be correct when there is a TNP change made.  There is no way to detect that a PropertyString value needs to be changed.

So, I think the right approach here is to see if those can change type to be Links, at which point the TNP code should cut in and automagically make the right thing happen.

maxwxyz, 2024-06-03:
Will this be solved until 1.0? Not for feature freeze but bugfix?

PaddleStroke, 2024-06-03:
Yes I hope so.

bgbsww, 2024-06-06:
I'm with @PaddleStroke here; it will get solved, just a question of collaborations and when.  If you need me just @

PaddleStroke, 2024-06-06:
@bgbsww Can you please elaborate in more details?
Or alternatively maybe we can chat a little bit? if you can send me you availability on the forum via MP for example. (my forum pseudo is paddle)

bgbsww, 2024-06-06:
I'm constantly on Telegram and discord, bgbsww both places, usually 8a to 8p UTC-04.

So, if you look in src/App/PropertyLinks.cpp, you'll see that starting in the PropertyLinkBase class and working up through all the other PropertyLink classes, they are now set up to adjust a link if the underlying object changes.

In assembly, you are linking to objects by using a PropertyString, which is just a name and cannot adapt if the underlying object changes.  I'm suggesting that for the Element1, Element2, Vertex1 and Vertex2 properties you use a PropertyLink.

I just tried flipping those ( and Object1 and Object2 ) out in createProperties method, and I now see a need for the proper lookups and assignment in setJointConnectors.  I'll play with that a little more.

bgbsww, 2024-06-07:
Okay, let's try that differently now that I'm in there:  Leave everything alone, except:
diff --git a/src/Mod/Assembly/JointObject.py b/src/Mod/Assembly/JointObject.py
index f7cd18a122..0ac0679402 100644
--- a/src/Mod/Assembly/JointObject.py
+++ b/src/Mod/Assembly/JointObject.py
@@ -195,7 +195,7 @@ class Joint:
         # First Joint Connector
         if not hasattr(joint, "Object1"):
             joint.addProperty(
-                "App::PropertyString",  # Not PropertyLink because they don't support external objects
+                "App::PropertyLink",  # Not PropertyLink because they don't support external objects
                 "Object1",
                 "Joint Connector 1",
                 QT_TRANSLATE_NOOP("App::Property", "The first object of the joint"),
@@ -250,7 +250,7 @@ class Joint:
         # Second Joint Connector
         if not hasattr(joint, "Object2"):
             joint.addProperty(
-                "App::PropertyString",
+                "App::PropertyLink",
                 "Object2",
                 "Joint Connector 2",
                 QT_TRANSLATE_NOOP("App::Property", "The second object of the joint"),
@@ -470,7 +470,7 @@ class Joint:
         isAssembly = assembly.Type == "Assembly"
 
         if len(current_selection) >= 1:
-            joint.Object1 = current_selection[0]["object"].Name
+            joint.Object1 = current_selection[0]["object"] #.Name
             joint.Part1 = current_selection[0]["part"]
             joint.Element1 = current_selection[0]["element_name"]
             joint.Vertex1 = current_selection[0]["vertex_name"]
@@ -478,7 +478,7 @@ class Joint:
                 joint, joint.Object1, joint.Part1, joint.Element1, joint.Vertex1
             )
         else:
-            joint.Object1 = ""
+            joint.Object1 = None
             joint.Part1 = None
             joint.Element1 = ""
             joint.Vertex1 = ""
@@ -486,7 +486,7 @@ class Joint:
             self.partMovedByPresolved = None
 
         if len(current_selection) >= 2:
-            joint.Object2 = current_selection[1]["object"].Name
+            joint.Object2 = current_selection[1]["object"] #.Name
             joint.Part2 = current_selection[1]["part"]
             joint.Element2 = current_selection[1]["element_name"]
             joint.Vertex2 = current_selection[1]["vertex_name"]
@@ -504,7 +504,7 @@ class Joint:
                 self.updateJCSPlacements(joint)
 
         else:
-            joint.Object2 = ""
+            joint.Object2 = None
             joint.Part2 = None
             joint.Element2 = ""
             joint.Vertex2 = ""

PaddleStroke, 2024-06-07:
Object1/2 have to be strings actually. The reason is as it's writen in the comment:  `# Not PropertyLink because they don't support external objects`
So if you insert an external object in your assembly then it won't work. This is why I had to change from PropertyLink to PropertyString for those.

But anyway object1/2 here should not generate any TNP problem anyway. The issue is with element/vertex properties no?

bgbsww, 2024-06-07:
PropertyLinkSub PropertyXLink both have code that supports external objects now.  I'm not sure about PropertyLink. 
 Without a deeper understanding of assembly I can't really say which properties need to change, but I would tend to expect all of them.

The subobject links come into play when you have a sketch as a base for a pad or other extruding operation, and you change the sketch - you'll have the same object, but with different subobjects.  The object links may come into play when you have a boolean or other combination of objects that are referenced.  Testing would clarify this.

TNP fixes are not going to play well with object referencing properties in strings as there is nothing to tell TNP mitigation to activate.

In my exploration yesterday, it was clear that if you do end up changing the types of the element/vertex properties, you'll need to change the places that assign to them as well to provide subobject links and not just strings.

PaddleStroke, 2024-06-10:
PropertyLink are a bit messy, there are so many types, and not much documentation. What is the difference between PropertyXLink and PropertyXLinkSub ? Apparantly PropertyXLink also store the sub so who knows.

And between PropertyXLink and PropertyLinkGlobal? It's not very clear.

I will try to replace Object1 + Element1 + Vertex1 all by one PropertyXLink and see what happens.

bgbsww, 2024-06-10:
I certainly agree, they are messy, and need at least some documentation.  I
wish I understood them better.  I just understand the TNP awareness, not
the full depth.

bgbsww, 2024-06-10:
I'm 90% sure that's the case ( there is code elsewhere that does that ).  If the TNP code didn't throw so many surprises, I'd be sure.

PaddleStroke, 2024-06-10:
Apparantly it doesn't.
I created the joint using face12, then change the shape, the face becomes face5. The PropertyXLinkSub still displays face12.

PaddleStroke, 2024-06-13:
@bgbsww any suggestion about it? I'm not sure what to do.

@chennes you seemed involved in TNP as well, do you per chance have an idea of what I need to do ?

What I tried so far : 
in a PropertyXLink or a PropertyXLinkSub, I store the object and subnames like so 
`joint.Object1 = [obj, elName]`

This seems to work, the property can be seen in property manager. However if a TNP event occurs and the element name changed, the property is not modified and still holds the old element name.

PaddleStroke, 2024-06-14:
I have tested this in Linkstage and confirm that it works there. The element of a PropertyXLinkSub update when the shape changes.

But in current main it does not work yet.

On my end I will make the change in assembly to use PropertyXLinkSub. So when it will be fixed in main TNP will work in assembly.

bgbsww, 2024-06-14:
That sounds correct to me.  I will look for the PropertyXLinkSub issues in
a bit - it is about 4 bugs away for me right now.

closed 2024-07-01 by chennes