FreeCAD/FreeCAD #14105: Modelling with context of an assembly in Part Design

drwho495, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

# The problem
When modeling in CAD software, parts are usually assembled on an assembly workbench or file. You can usually edit a model in context of the assembly, so the sketches of a part can have a reference to the assembly that the part is in. This feature is not available on FreeCAD in any of the assembly workbenches. 

# My Solution
I have a possible solution for this problem. I currently have a FreeCAD fork with the option of merging into the master branch of FreeCAD. App images are available, but there are some example videos that I have created to show how this tool works.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37083 (Git) AppImage
Build type: Release
Branch: main
Hash: 502f78e111c761cb92c51222605b26b8701fcbca
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.6
  * Manipulator 1.5.7
  * OpenDark 2023.12.17
  * AutoSave
  * fasteners 0.5.13
  * Assembly4 0.50.6
  * QuickMeasure 2022.10.28
  * freecad.gears 1.2.0
  * Ondsel-Lens 2024.1.22.01
  * Assembly3 0.12.0
  * Render 2024.1.23
  * A2plus 0.4.61
  * Help 1.0.3
  * 3D_Printing_Tools
  * kicadStepUpMod 10.23.2
```

### Subproject(s) affected?

Assembly

### Anything else?

https://github.com/drwho495/FreeCAD-Context-Bundle
https://github.com/drwho495/FreeCAD-Context-Fork

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
This looks cool! If you are done please create a PR. To be considered for a FreeCAD 1.0 release the PR has to be submitted until Monday, 2024-06-03 latest (status ready for review)
@PaddleStroke FYI

drwho495, 2024-05-17:
I'm just going to work on some bug fixes and adding try-catches to stop errors, I should have it done by the end of the day.

PaddleStroke, 2024-05-17:
A video of the feature would be great to better understand

maxwxyz, 2024-05-17:
@PaddleStroke 
[image]

drwho495, 2024-05-17:
> A video of the feature would be great to better understand

I am planning on making one soon.

PaddleStroke, 2024-05-17:
@maxwxyz thanks but without audio explanation I can't understand what's happening in the gif

maxwxyz, 2024-05-17:
@PaddleStroke the positions of the holes in the base pad is referenced from the position in the assembly file. when the assembly is moved the holes in the part design body can be updated.

drwho495, 2024-05-18:
Ok, the PR is up.