FreeCAD/FreeCAD #14107: Sketcher: Preferences have no default value

maxwxyz, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Sketcher preferences for OVP and Dimension tool have no default value. Nothing to fall back if a user clicks on "reset":
[image]

These preferences seem to use a `QComboBox` which is populated through the .cpp file and the QComboBox has no member for `onRestore()`.
Maybe it should use a `PrefComboBox` with `stdset="0"` if this is used for resetting the preferences.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CfdOF 1.25.13
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-17:
@PaddleStroke @kadet1090 FYI

PaddleStroke, 2024-07-03:
The defaults are defined in the cpp file : 

```
    bool singleTool = hGrp->GetBool("SingleDimensioningTool", true);
    bool SeparatedTools = hGrp->GetBool("SeparatedDimensioningTools", false);
```

So if you do reset all it will remove the parameters which in turn will use the defaults as set there.
However if you reset only the page or the group then I don't know how its implemented so I'm not sure what's happening here.

PaddleStroke, 2024-07-03:
Ok I found an example, it needs to implement resetSettingsToDefaults in the sketcher pages.

closed 2024-07-15 by chennes