FreeCAD/FreeCAD #14110: Sketcher UI: Horizontal / Vertical constraints rotate with view

maxwxyz, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When applying horizontal and vertical constraints they are displayed correctly when the sketch is viewed in the direction of the sketch plane. Same as any constraints:
[image]

When you rotate the view, all constraints stay aligned to the view so the user can read dimensions and all constraint icons. That is ideal.
The only issue is with the horizontal and vertical constraints. They also stay oriented the same way to be identified as horizontal and vertical, but they look confusing when the sketch is rotated or viewed from any other angle.

[image]

When the view is rotated 90 degrees the horizontal ones appear vertical next to a horizontal appearing line:
[image]

As these are the only constraints which are relative to the sketch orientation (axes/grid) shouldn't they be displayed always oriented aligned with the horizontal and vertical axes? CATIA for example just uses a H and V for indication.

@FreeCAD/design-working-group FYI

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-05-17:
Well, that is quite interesting. H and V seem like reasonable substitutes, and most likely are the simplest solution to this problem. I'd wait for others to weigh in, but my initial reaction is to support this.

pierreporte, 2024-05-17:
Even the Catia implementation isn’t ideal. When you’re applying an H or V constrain when the view is rotated, you’ll make the mistake half the time or more. My favourite solution so far is what Solid Edge does: drawing + crosses at the center of the line, in the sketch plane, aligned with the lines. It has two drawbacks: it needs merged H/V constrains (which wasn’t very popular in the FreeCAD community) and competes with midpoint. Pictures below.

[image]

[image]

[image]

[image]

pierreporte, 2024-05-17:
I would say that, if the Solid Edge solution was to be implemented in some way, the cross could be displayed between all points lying on the line. If there is no intersection with the segment (except at end points), there is only one cross. If there is one intersection, there are two crosses, etc.

maxwxyz, 2024-05-17:
Don't forget that you can also only align points or centerpoints horizontally. Not sure if a cross in the middle of nothing would be ideal not to confuse with symmetry. Also with points only, the rotating aspect of the horizontal and vertical constraint makes even less sense.
Normal orientation:
[image]

90 degrees:
[image]

pierreporte, 2024-05-17:
Solid Edge draws a dashed line between points, and puts the cross on it. FreeCAD could do it in any case, it would add clarity. The thin green line system used for arcs could be used for this.

[image]

-----------------

NX has a similar solution: there is a thick dash on the line. It’s more distinguishable from points.

[image]

0penBrain, 2024-05-18:
First question is : why would you rotate your sketch ???
Using 'H' and 'V' is not better has you'll have an 'H' close to a vertical line. I would even say current system is better because you can more easily detect you're not in the genuine orientation if you come to such a situation. ;)

obelisk79, 2024-05-18:
> First question is : why would you rotate your sketch ??? 

I had this same thought, but my assumption is that a large number of CAD users like to model (and sketch presumably) in-situ. I don't think any of the other proposals above other than the H and V offer anything particularly useful, and even the H/V are only marginally useful, but they are also not really harmful in any way.

0penBrain, 2024-05-18:
> but that change would also not really be harmful in any way.

AFAIK symbols are pixmaps and thus not translatable. Using "hardcoded" H/V doesn't sound very good regarding i18n.

obelisk79, 2024-05-18:
> AFAIK symbols are pixmaps and thus not translatable. Using "hardcoded" H/V doesn't sound very good regarding i18n.

i18n is of course important, but in this case I think that enough major languages (nearly all latin/germanic) would see direct correlation, and for those which don't the letters merely become discrete symbols to learn and these aren't actual words in context here which would need direct translation. Out of curiosity I punched the words vertical and horizontal into google translate and found many (some surprising, indonesian!?!) languages where H and V would directly apply.

In either case, I have no issues with leaving behavior as it currently stand. Even if a user wants to rotate a sketch or skew to a 3d perspective, it is a very poor practice (my opinion) to try and sketch from such a perspective.

pierreporte, 2024-08-13:
Rotating sketch is often needed when working with external geometry that is out of plane, or to check how it looks with other parts nearby. Though it’s never rotating the sketch at 90°—that often happens by accident.