FreeCAD/FreeCAD #14117: PartDesign Mirrored

free777cat, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PartDesign Mirrored
It is impossible to leave only the mirrored part.

Only the mirrored part is needed for further actions.

### Full version info

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * FEMbyGEN 2.1.0
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * sheetmetal_середа, 10 квітня 2024.zip
  * fasteners 0.4.72
  * sheetmetal_субота, 06 квітня 2024.zip
  * sheetmetal.zip
  * lattice2 1.0.0
  * FEM_FrontISTR 0.1.0
  * sheetmetal_work_dxf.zip
  * sheetmetal_субота, 30 грудня 2023.zip
  * SheetMetalFoldCmd.py_повністю_робоча_неділя, 21 квітня 2024.zip
  * DynamicData 2.56.0
  * sheetmetal 0.4.13
  * Defeaturing 1.2.2
  * dxf-library
  * OpticsWorkbench 1.0.12
  * temp
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
related: https://github.com/FreeCAD/FreeCAD/issues/5678
You can just use the mirror tool in Part WB as workaround.
Please be a bit more precise what your problem is: do you want to mirror a single or multiple features and only keep the mirrored copies or are you trying to mirror the entire body and keep only the mirrored body?

free777cat, 2024-05-18:
Task: mirror the whole body and save only the mirror body.

maxwxyz, 2024-05-18:
that's possible either with the PartWB mirror or scale (-1).

free777cat, 2024-05-18:
"Part WB" does not provide all necessary features.

maxwxyz, 2024-05-18:
Then please describe the problem precisely in the original issue.

free777cat, 2024-05-18:
The possibility of this option in "PartDesign" seems logical.
Also "PartDesign" goes better with "SheetMetal". Which is very important.

maxwxyz, 2024-05-18:
Yes, but this would be another body (a mirrored clone) and not a feature inside a body, right?

free777cat, 2024-05-18:
In the future, I conduct an assembly from files created in this way: "example".
The mirrored part must be in the same "body".
It will be able to change with the introduction of changes to the previous "history".
[example.zip](https://github.com/FreeCAD/FreeCAD/files/15362511/example.zip)

free777cat, 2024-05-18:
Зберігаються усі "привязки" до LCS.

free777cat, 2024-05-19:
Such an option would make "PartDesign" more versatile. Are changes being considered?

free777cat, 2024-05-27:
> Yes, but this would be another body (a mirrored clone) and not a feature inside a body, right?

This is not a new "body". If we need to quickly change the shape to mirror?
It will also significantly speed up the design process. I cannot understand the reason for refusing to add it.