FreeCAD/FreeCAD #14119: Std_Measure: Label is editable but isn't

0penBrain, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * New document
 * Add any object, any measurement (validate & close Measure tool)
 * Select measurement in the tree view
 * Try to edit Label in the Property Pane => :warning: Label is editable and even Expression-aware, but any change will be refused

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37361 +1 (Git)
Build type: Debug
Branch: myBranch
Hash: 761d8fc69d0d9a67d0a6cd4da8fdb35401c0a7e6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Anything else?

With multiline label (such as Center Of Mass), it even triggers errors in the Report View when editing

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
@hlorus FYI can confirm

FEA-eng, 2024-06-29:
I can't reproduce, the name (not the value of the measurement) part of the label can be changed. Could you check in the newest weekly?

0penBrain, 2024-06-29:
@FEA-eng Problem is partly solved. The Label editing works correctly.
But still this field is expression-aware, which is a problem as it generates errors if you enter a valid expression.

luzpaz, 2024-07-22:
Whats the solution here ?

0penBrain, 2024-07-22:
> Whats the solution here ?

Well. I did more tests and actually this isn't related to Measure but is FC wide.
I see 3 solutions:
 * Force getting expression return as a string (I don't know if it's currently supported)
 * Disable expression in Label (but probably this is a feature regression)
 * Keep as is and bet on the user to do the conversion to string

Anyway, probably this bug can be closed and eventually a new one opened.

benj5378, 2024-10-04:
The label is editable, and the edit is saved, as long as the first word is changed:

Change accepted: Length: 10,00 mm -> Length of box: 10,00 mm
Change denied: Length: 10,00 mm -> Length: 1124525,00 mm

I am not sure this is actually a bug? I don't see the unexpected behaviour? However, the expression icon should be disabled.

maxwxyz, 2024-10-05:
iirc anything before `:` can be edited. Maybe only this part should be presented if the user wants to rename it. The value updates if the measurement changes so it always reflects the true measurement.

FEA-eng, 2025-06-27:
Still relevant

0penBrain, 2025-07-11:
@FEA-eng I can't reproduce with latest weekly. Could you double check please ?

FEA-eng, 2025-07-11:
> [@FEA-eng](https://github.com/FEA-eng) I can't reproduce with latest weekly. Could you double check please ?

I meant that what @benj5378 said is still relevant - only the part before `:` can be edited (that's unlikely to be a bug though) and there's the expression button that probably shouldn't be there.

JustTryingToGetSomeWorkDone, 2026-09-10:
In 26.3.0dev 2026/09/02 - Still an issue. I tried to rename a measurement to indicate what the measurement was targeting, and it discards any changes made.

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/92013bba9e782aa32731565d24603d2aaa7da24d

Renaming multiline Position and COM labels no longer generates an invalid Python command. This does not change the expression button or make the measured value editable.