FreeCAD/FreeCAD #14121: BIM: Preferences group is still called "Arch"

Roy-043, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is still an "Arch" group in the preferences. This should be renamed to "BIM".

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
@yorikvanhavre FYI

yorikvanhavre, 2024-05-21:
Indeed. I'll submit a PR for that.

PR #14188 (2024-05-22): BIM: Reworked preferences and setup screens
* Reworded welcome screen, removed "update" section
* Removed Webtools and Start option from setup dialog - fixes #14179
* Renamed Arch preferecnes to BIM - fixes #14121
* Moved NativeIFC preferences to BIM - fixes #14137
commit d28a7971 (2024-05-22): BIM: Reworked preferences and setup screens

* Reworded welcome screen, removed "update" section
* Removed Webtools and Start option from setup dialog - fixes #14179
* Renamed Arch preferecnes to BIM - fixes #14121
* Moved NativeIFC preferences to BIM - fixes #14137
closed 2024-05-22 by yorikvanhavre