FreeCAD/FreeCAD #14122: [Feature Request] Assembly: Pattern for imported objects

pure-machinery, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The most obvious problem arises with mechanical elements such as bolts. When building up the assembly it can be found to have hundreds of bolts and washers. Inserting the standard parts one by one is tedious which a pattern command will solve. The pattern command should also be pattern-aware - meaning If an imported part has a pattern feature (e.g. linear or polar) and another part is constrained to this feature in some way, once you pattern the second part it should follow the first parts pattern feature. 

This is convenient because if you want to have less or more patterned features all you have to do is decrease the number of occurrences in the base part, and later refresh the assembly.

I'm sure this conversation occurred somewhere but I could not find it on GitHub or the forums.  

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BillOfMaterials 0.0.8.2
  * sheetmetal 0.4.13
  * fasteners 0.5.20
  * A2plus 0.4.64a
  * fcgear 1.0.0
  * OpenTheme 2024.5.3
  * IconThemes
```

### Subproject(s) affected?

Assembly

### Anything else?

Here's an old video for PTC Creo: [https://www.youtube.com/watch?v=cH-ONgCG9e0](https://www.youtube.com/watch?v=cH-ONgCG9e0)

And CATIA: [https://www.youtube.com/watch?v=etK7yv-_t8Y](https://www.youtube.com/watch?v=etK7yv-_t8Y)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
@PaddleStroke FYI
As workaround a value like "insert x times" would be helpful.

EternityForest, 2024-06-04:
Realthunder lets you pattern basically anything, including another pattern, and patterns can result in multiple solids.

I wonder if his change can be used?

pierreporte, 2024-08-28:
It would be necessary that the BOM tool understands that there is a pattern and computes the quantity accordingly.

eblanshey, 2026-10-01:
Does the new patterning capability allow patterning parts based on existing patterns made in Part/PD, such as holes, as this issue describes? When I watched Paddlestroke's demo video a few weeks ago I only saw circular, path, and point, but not specifying another pattern feature.

maxwxyz, 2026-10-01:
@eblanshey for Part Design: new patterns along the existing ones: circular (concentric circles), path (along a path object incl. offsets), and point. They work like the other PD patterns: select feature(s) or pattern the entire body.
Assembly (does not need an assembly, also works in Part): Adds new LinkArrays to pattern objects as links. Types are: Linear, Polar, Circular, Path, Point. Linear needs a direction (can be an axis or edge) and supports 2 directions (matrix-like). Polar requires an axis. path an object with wires (e.g. a sketch) or select an individual edge. Point neds a point object, e.g. a sketch with points or it uses points from the sketch itself (e.g. polyline).

Both tools allow to suppress instances, including the option to suppress/hide the original feature/object.

You can try it already: https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2026.10.01

pure-machinery, 2026-10-01:
> Does the new patterning capability allow patterning parts based on existing patterns made in Part/PD, such as holes, as this issue describes? When I watched Paddlestroke's demo video a few weeks ago I saw circular, path, and point, but not specifying another pattern feature.

Yup, it does not close the issue. 

I won't downgrade the features PaddleStroke build, since they are great :). 

But I would wager that this issue is still as important. Also I have already commented on the original PR that it didn't fix this issue.

PaddleStroke, 2026-10-02:
Can you guys give me a video showing the use case you are describing because I am not sure how to interpret it exactly?

pure-machinery, 2026-10-02:
@PaddleStroke Here's one I have found on YouTube (PTC Creo):

[https://www.youtube.com/watch?v=D9hlxhaH7es](url)

Keep in mind this applies to the patterned bodies in Assembly if they have a constraint to the pattern feature geometry. Kind of like if I was to place a bolt inside a patterned hole, and later pattern the bolt - only by reference of the hole pattern.

eblanshey, 2026-10-02:
Concrete steps you would take in FC, as seen in the videos @pure-machinery shared:

1. Create base pattern: in PartDesign pattern a hole
2. In Assembly workbench insert a bolt and join it to one of the holes (it can be the original hole, or one of the patterned holes)
3. Select the bolt, run "Pattern by Reference Pattern" command.
4. The task panel opens to show you the pattern feature reference that is detected from the bolt's initial joint references. The preview shows the bolts that would be placed
5. Clicking OK inserts more instances of the bolt

If made to be properly parametric, altering the first pattern (shape, number of instances) would automatically update the positioning and number of instances of the bolt in the assembly. Perhaps that could be a followup feature if too complex.

One optional convenience feature to do the whole process in less steps: step number 2 above, to join the first bolt, could auto-detect that the selected joint references is part of a pattern, and offer a checkbox to "pattern part using reference pattern". That would allow you to skip step 3, 4, and 5.

@pure-machinery you may want to rename this issue to "Assembly: pattern using reference pattern".

maxwxyz, 2026-11-07:
cacciatorino added a [flange and bolt example](https://github.com/user-attachments/assets/6157b662-89af-4db3-ac61-bf7f32c38194) in #33923: place the first bolt, then have the others follow the flange's hole pattern, including changes to hole positions and count.