FreeCAD/FreeCAD #14123: BIM: "Single IFC document" pop-up displays after every document change

Roy-043, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Test:
1. Switch to the BIM WB.
2. Create a document.
3. Click "No" in the "Single IFC document" pop-up.
4. Create another document.
5. Again click "No" in the pop-up.
6. Switch between the documents.
7. Result: pop-up is displayed after every switch.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-21:
Indeed the current process is not good. I think the best course of action for now is to disable this "automatic" system entirely, and only propose it when the user explicitly presses the button to convert a FreeCAD document to NativeIFC. In other words, remove the ability to always start in NativeIFC mode.

This functionality is still important, but the current implementation is bad and should be redone. I'lll submit a PR to remove the current implementation

Roy-043, 2024-06-28:
Cannot reproduce the issue with the version listed below.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
```

closed 2024-06-28 by Roy-043