FreeCAD/FreeCAD #14126: Sketcher: Polyline has inconsistent behavior when clicking on existing endpoints

0penBrain, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Empty sketch, Polyline tool
 * Draw a triangle using 3 lines (default mode) => :warning: Notice on the last click (closing the shape), tool terminates the current polyline drawing
 * Click somewhere in the void and set the 2nd endpoint on one already existing => :warning: This time on the last click the tool isn't terminating and continue drawing the polyline

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37361 +1 (Git)
Build type: Debug
Branch: myBranch
Hash: 761d8fc69d0d9a67d0a6cd4da8fdb35401c0a7e6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2025-09-14:
Still valid

FEA-eng, 2025-09-14:
Strange behavior, I can confirm it too.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-11-14:
Thanks @PaddleStroke!  
When the time comes could you reference this ticket in the PR ?

luzpaz, 2026-10-10:
@PaddleStroke ping