FreeCAD/FreeCAD #14128: Sketcher: Unhandled exception in case of setting coincidence on both endpoints of a line/arc

0penBrain, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Empty sketch, draw a single line
 * Select both endpoints and set a coincident constraint => :warning: The constraint is created and errors are triggered in the Report View

Same exists with Arc instead of Line, but it may be more weird as constraint may or not trigger the error immediately. If not, it will create them when trying to move the center/endpoints.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37361 +1 (Git)
Build type: Debug
Branch: myBranch
Hash: 761d8fc69d0d9a67d0a6cd4da8fdb35401c0a7e6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-05-18:
To solve the line issue the function SketchObjectPy::addConstraint could check the return value of solve(). If the value is -1 it simply removes the added constraint.

Another possibility could be to implement a function that checks invalid coincidences beforehand and aborts the operation. E.g. coincidences are invalid if the two points are from the same geometry -- except for splines.

0penBrain, 2025-09-14:
Still valid

FEA-eng, 2025-09-14:
Pretty nasty bug because even though you're not supposed to apply a constraint this way, the "Solver failed to converge" behavior basically breaks the entire sketch.

FEA-eng, 2025-11-14:
Still "Solver failed to converge" and these errors:

```
09:38:00  Updating geometry: Error build geometry(0): Both points are equal
09:38:00  Invalid solution from DogLeg solver.
09:38:00  Updating geometry: Error build geometry(0): Both points are equal
09:38:00  Invalid solution from LevenbergMarquardt solver.
09:38:00  Updating geometry: Error build geometry(0): Both points are equal
09:38:00  Invalid solution from BFGS solver.
```

@PaddleStroke Maybe this could be prevented somehow.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

PaddleStroke, 2025-11-14:
Note this does not happen if you first select the tool, then the points. So it appears the tool is already preventing that in this workflow. It just doesn't in the other workflow points->tool

PaddleStroke, 2025-11-14:
Interestingly, the current tool->point has the GeoId1 != GeoId2 check, but it does not check if GeoId1 is a bspline. So currently you cannot make a coincidence between the start/end of a bspline with the tool->point workflow. I'll fix that too.

closed 2025-11-24 by chennes