FreeCAD/FreeCAD #14129: Problem:  Toponaming fails with index error

sliptonic, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

testfile attached.   rename tntest.zip to tntest.fcstd

This is a straightforward test case.  Open the sketch 'stretchme'  and drag the right edge over.

[image]
[tntest.zip](https://github.com/FreeCAD/FreeCAD/files/15365738/tntest.zip)

I see the following in the console
```
18:05:04  <PropertyLinks> PropertyLinks.cpp(453): tntest#Sketch002.Support missing element reference tntest#Pocket ;g11;SKT;:H1db,E;FAC;:H1db:4,F;:G0;XTR;:H1db:8,F;:H-1de,F.Face11
18:05:04  Sketch002: Shape index 11 out of bound 10
18:05:04  Sketch002: Shape index 11 out of bound 10
```

### Full version info

```shell
OS: Linux Mint 21 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37360 (Git)
Build type: Unknown
Branch: main
Hash: 2d2977abafd93c9ab97f9ad7beb25f5311c50b06
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.15
  * BillOfMaterials 0.0.7.2
  * sheetmetal 0.4.9
  * Ondsel-Lens 2024.1.15.01
  * FeedsAndSpeeds 0.5.0
```

### Subproject(s) affected?

None

### Anything else?

Compiled with toponaming enabled.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
@bgbsww FYI

bgbsww, 2024-05-20:
So, after cleaning up other issues, we now seem to be matching the LS3 behavior.  In both cases, if you stretch the edge of the sketch a pocket is based on to outside of the supporting face, it doesn't redraw, you get console errors, and if selected, the selection shows a history name:

LS3:
16:09:29  <App> PropertyLinks.cpp(439): tntest#Sketch002.Support missing element reference tntest#Pocket ;#60;:H-1de,F.Face11
16:09:29  6.5e-08 <App> Document.cpp(3424): Failed to recompute tntest#Sketch002: AttachEngine3D: subshape not found Pocket.Face11
16:09:29  Recompute failed!

Main:
16:07:55  <PropertyLinks> PropertyLinks.cpp(453): tntest#Sketch002.Support missing element reference tntest#Pocket ;g11;SKT;:H1db,E;FAC;:H1db:4,F;:G0;XTR;:H1db:8,F;:H-1de,F.Face11
16:07:55  Sketch002: Shape index 11 out of bound 10
16:07:55  Sketch002: Shape index 11 out of bound 10

And in both cases, moving the pocket edge back in bounds works and seems to clean the maps correctly.  This is after #14178 is in place.

bgbsww, 2024-07-06:
Is there a reason this isn't closed?  Or at least removed from the 1.0 and blocker lists?  Behavior should be matching LS3 now, so anything else is essentially a new feature.

maxwxyz, 2024-07-12:
Missed that, thanks.

drwho495, 2025-07-19:
.

drwho495, 2026-03-26:
> I opened this, and FreeCAD crashed immediately. Tried once more, and it opened.
> The issue is still happening.
> 
> Is this something that can be fixed somehow?
> When the Pocket changes, the number of faces produced changes too. 
> And since Sketch002 is attached to a missing face, how it can be mapped to a face that now has a different Name?
> 
> ```
> OS: Ubuntu 25.10 (ubuntu:GNOME/ubuntu/wayland)
> Architecture: x86_64
> Version: 1.2.0dev.20260324 (Git shallow)
> Build date: 2026/03/24 17:14:33
> Build type: Release
> Branch: (HEAD detached at 1bc5980a7)
> Hash: 1bc5980a7ff302a55c617181e774cd5cd2a9443a
> Python 3.11.14, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
> Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
> Logical DPI/Physical DPI/Pixel Ratio: 96/144.289/1
> ```

This is an old file from 0.22dev, so crashes are probably normal. The issue is still present with the current toponaming algorithm, but should be fixed with my V2 toponaming algorithm. You can try it early [here](https://GitHub.com/drwho495/freecad/releases)