FreeCAD/FreeCAD #14130: FEM: Unidentified CalculiX error when applying a rigid body constraint to an edge/vertex

FEA-eng, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Normally, rigid body constraint is applied to faces. But it is (and should be) possible to apply it to edges and vertices as well. However, when doing so, I get a strange CalculiX error:

`CalculiX finished with error 4294967295.`

The whole output looks like this:

<details>
<summary>Click to expand</summary>

```
16:48:07  Run of CalxuliX ccx tools solver started.
16:48:07  
16:48:07  Check prerequisites...
16:48:07  
16:48:07  Get mesh data for constraints, materials and element geometry...
16:48:07  [{'ccx_elset': 'Evolumes', 'ccx_elset_name': 'MaterialSolidSolid', 'mat_obj_name': 'MaterialSolid', 'ccx_mat_name': 'AlMg3F24'}]
16:48:07  ConstraintFixed:
16:48:07      Type: Fem::ConstraintFixed, Name: ConstraintFixed
16:48:07      ReferenceShape ... Type: Face, Object name: Box, Object label: Cube, Element name: Face5
16:48:07  ConstraintRigidBody:
16:48:07      Type: Fem::ConstraintRigidBody, Name: ConstraintRigidBody
16:48:07      ReferenceShape ... Type: Edge, Object name: Box, Object label: Cube, Element name: Edge2
16:48:07  Getting mesh data time: 0.031 seconds.
16:48:07  
16:48:07  CalculiX solver input writing...
16:48:07  Input file:C:\Users\GAZ87HD3\AppData\Local\Temp\fcfem_o8ho9kl7\FEMMeshGmsh.inp
16:48:07  One monster input file.
16:48:07  Writing time CalculiX input file: 0.047 seconds.
16:48:07  
16:48:07  CalculiX solver run...
16:48:07  OverflowError
16:48:07  CalculiX failed with exit code 4294967295
16:48:07  --------start of stderr-------
16:48:07  --------end of stderr---------
16:48:07  --------start of stdout-------
16:48:07  
************************************************************

CalculiX Version 2.17, Copyright(C) 1998-2020 Guido Dhondt
CalculiX comes with ABSOLUTELY NO WARRANTY. This is free
software, and you are welcome to redistribute it under
certain conditions, see gpl.htm

************************************************************

You are using an executable made on Thu Jul 23 21:43:50 CEST 2020

  The numbers below are estimated upper bounds

  number of:

   nodes:         2221
   elements:         1849
   one-dimensional elements:            0
   two-dimensional elements:            0
   integration points per element:            4
   degrees of freedom per node:            3
   layers per element:            1

   distributed facial loads:            0
   distributed volumetric loads:            0
   concentrated loads:            1
   single point constraints:          738
   multiple point constraints:           40
   terms in all multiple point constraints:          235
   tie constraints:            0
   dependent nodes tied by cyclic constraints:            0
   dependent nodes in pre-tension constraints:            0

   sets:            6
   terms in all sets:         8097

   materials:            1
   constants per material and temperature:            2
   temperature points per material:            1
   plastic data points per material:            0

   orientations:            0
   amplitudes:            2
   data points in all amplitudes:            2
   print requests:            1
   transformations:            0
   property cards:            0

 STEP            1

 Static analysis was selected

 Decascading the MPC's

 Determining the structure of the matrix:
 number of equations
 5856
 number of nonzero lower triangular matrix elements
 197025

 Using up to 8 cpu(s) for the stress calculation.

 Using up to 8 cpu(s) for the symmetric stiffness/mass contributions.

 Factoring the system of equations using the symmetric spooles solver
 Using up to 8 cpu(s) for spooles.

16:48:07  
--------end of stdout---------
16:48:07  --------start problems---------
16:48:07  
--------end problems---------
16:48:07  CalculiX finished with error 4294967295.
16:48:08  Run of CalxuliX ccx tools solver finished.

```

</details>

I've noticed the `OverflowError` message at the top.

The input file is generated correctly, it works when I submit it using standalone CalculiX. Then I can even import it back to FreeCAD.

I've tried with the ccx binary provided with 0.22 dev as well (this one is 2.17 as it's taken from 0.21.2 stable) but the issue persists.

[RB_error.zip](https://github.com/FreeCAD/FreeCAD/files/15365755/RB_error.zip)

@marioalexis84 Do you know what might be causing this?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-05-20:
On my system there is no error but no result (.frd file is empty).
The CalculiX documentation says:
> If the participating nodes in a rigid body definition lie on a straight line,
the rigid body rotation about the line is not defined and an error will occur.
To remove the rotational degree of freedom, specify that the rotation about the
axis is zero

Effectively, removing the rotational degree of freedom generates the result.

FEA-eng, 2024-05-20:
@marioalexis84 Ok, right, thanks. I'll update the documentation then. The same seems to be true for vertices. I guess the issue can be closed then, right?

marioalexis84, 2024-05-20:
yes, it can be closed

FEA-eng, 2024-05-20:
Ok, closing. I just wonder why it worked with standalone CalculiX. Maybe this limitation was fixed without notice in the newest version or it’s another case of inconsistent behavior of CalculiX.

closed 2024-05-20 by FEA-eng

marioalexis84, 2024-05-20:
>I just wonder why it worked with standalone CalculiX

It's hard to say without debugging CalculiX properly. The documentation says "an error will occur" but is not very specific about the error.
I'm using calculix 2.21 and 2.20 on Linux and it doesn't work.
It seems to be some kind of undefined behavior: one system raises an error, another doesn't complain, and another fails without an error.