FreeCAD/FreeCAD #14131: [Problem] CAM, Pocket ignores Extend Corners

tcurdt, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When cutting a pocket it might be necessary to cut further into the corners as sharp corners are not possible with a round endmill. That's what the "Extensions" are for.

I have enabled "Extensions" and "Extend Corners" but the setting is just ignored. I would have expected the path to go further into corners.

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
```

### Subproject(s) affected?

CAM/Path

### Anything else?

If not reproducible I am happy to provide the FCStd file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/10802

closed 2024-05-18 by maxwxyz

tcurdt, 2024-05-18:
I thought I had already opened this but couldn't find it. 🙄
Sorry for the duplicate.