FreeCAD/FreeCAD #14133: FEM: Constraint rigid body should use bigger spheres for reference point symbols

FEA-eng, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sphere showing the location of the rigid body reference point should be bigger because it may get covered by the other symbols representing this constraint:

[image]

IMO it should be bigger than those spheres on the sticks.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
[image]

marioalexis84, 2024-05-19:
We can also use another symbol. Maybe three orthogonal crossed lines?
[image]

FEA-eng, 2024-05-19:
> We can also use another symbol. Maybe three orthogonal crossed lines?

Or a cross, maybe even with label like in Abaqus (just way bigger):

[image]

But I prefer a sphere. 

A classic spider as a way to represent the whole constraint would be ideal but probably also hard to implement:

[image]

PR #14158 (2024-05-27): Fem: Increase size of reference node symbol in rigid body constraint - fixes #14133
Fixes #14133
@FEA-eng The sticks are a little thinner and the node a little bigger.
commit 459a5563 (2024-05-27): Fem: Increase size of reference node symbol in rigid body constraint - fixes #14133
closed 2024-05-27 by chennes