FreeCAD/FreeCAD #14135: FEM: Pipeline DataOnPoint hides entire result

maxwxyz, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to get the value at a point the DataOnPoint filter hides the entire result pipeline.
I guess that is related that the last  filter is only shown.
Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/72e3100e-d861-4da0-928e-bf1d7751ba0c

@FEA-eng @marioalexis84 FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.13
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
Was changed with https://github.com/FreeCAD/FreeCAD/pull/13820
It makes sense for all other filters though!

FEA-eng, 2024-05-18:
I can confirm this. One more filter is affected though - the Line clip filter.

PR #14157 (2024-05-27): Fem: Don't hide input filter with DataAtPoint and DataAlongLine - fixes #14135
Fixes #14135
commit 0041ec22 (2024-05-27): Fem: Don't hide input filter with DataAtPoint and DataAlongLine - fixes #14135
closed 2024-05-27 by chennes