FreeCAD/FreeCAD #14139: Manipulation of (built-in) Assembly objects fails with Gesture navigation

johnsonm, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have found no combination of modifiers by which I can manipulate Assembly objects under constraint with Gesture navigation (my default). If I switch to Touchpad, I can move assembled parts under constraint.

I have not surveyed all of the navigation modes to determine the full set that are compatible with the built-in Assembly workbench, only failed with Gesture and succeeded with Touchpad so far.

If it's only intended to work with certain navigation modes, documentation of that limitation and closing this report as won't fix or not a bug is fine by me.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37363 (Git)
Build type: Unknown
Branch: main
Hash: bfbe28f7a5e8098f1fe7239cda39697650027ff2
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
(Built with `FC_USE_TNP_FIX`)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-18:
@PaddleStroke FYI

luzpaz, 2024-08-18:
@Rexbas  any idea why this would happen ?

Rexbas, 2024-08-18:
So if I understand the issue correctly... You have an assembly, e.g. AssemblyExample.FCStd, then you double click Assembly in the tree view to get in the mode where you can manipulate the assembly. And then the problem is clicking somewhere on a part of the assembly and moving the mouse does not do anything with Gesture navigation style?

I quickly tested all navigation styles and here are my results:
Gesture works with ALT key but it should probably also work without it so this is a bug.
OpenInventor works with SHIFT key, this is by design.
OpenSCAD is bugged with LMB. It sometimes triggers both assembly manipulation and camera rotation, but it works when using the SHIFT key.

I also noticed that for Gesture the rotation mode as pictured in the tool tip ALT + LMB only works for right ALT not for left ALT.

All issues described here can probably be fixed in the navigation styles. I will add it to my long TODO list :smiley: .

johnsonm, 2024-09-20:
Thank you!

I definitely only used the left ALT and it never occurred to me to try right ALT. I'm right-handed, and the point of using Gesture for me is primarily that I almost exclusively use a touchpad for FreeCAD, so I want to drive the touchpad mostly with my dominant right hand, and use my left hand for modifiers. This means I have no recollection of ever using right CTRL, and recall using left ALT only by typing it to start compose sequences.

In the meantime, this hint will allow me to work in assembly without changing nav style back and forth a lot, until such time as it bubbles up to the top of your TODO list. :grin:

sslupsky, 2024-10-04:
I agree, it is preferable that you do not have to use the ALT key to select and move an object in Assembly WB.

@johnsonm Either the left or right ALT key works for me (I am running RC2 on macOS).

Rexbas, 2024-11-03:
I think this cannot be fixed before 1.0. It seems there is a conflict. The OpenSCAD and Gesture navigation style both rotate when pressing LMB. But LMB is also used for assembly manipulation. The navigation styles process the event first and so it never reaches assembly and I can currently not think of a good easy way to temporarily overrule the navigation style behavior from the assembly view provider.

I do have an idea but that is a massive refactor that also involves other navigation issues/feature requests, not going to happen anytime soon (by me at least).