FreeCAD/FreeCAD #14144: Tree view context menu "Send to Python Console" gone

krushia, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Not sure if intentional, but there is no longer a "Send to Python Console" in the tree view right-click menu. It was there in the previous AppImage.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-19:
It's still there in the latest weekly from yesterday:
[image]

krushia, 2024-05-19:
No its not.
[image]

krushia, 2024-05-19:
Apparently its missing when in the Part Design workbench?

krushia, 2024-05-19:
Same issue with Freeze, as described here. https://github.com/FreeCAD/FreeCAD/pull/12580#issuecomment-1978531015

maxwxyz, 2024-05-19:
Yes that's a known limitation but could be fixed easily. I'll open the issue again as now it's clear that you meant it's missing in Part Design. The freeze feature has some quirks for now: https://github.com/FreeCAD/FreeCAD/pull/12856

closed 2024-12-06 by chennes