FreeCAD/FreeCAD #14146: Expression access to "children" differs between container objects

krushia, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Writing an expression to access a child of a container is over-complicated because containers use different properties to list children. All of the following objects use an `App::PropertyLinkList` with a different name for grouping children.

The obvious containers:
- `Group` in `App::DocumentObjectGroup`, `App::Part`
- `ElementList` in `App::LinkGroup`

Some not so obvious containers:
- `Group` in `PartDesign::Body`
- `Links` in `Part::Compound`
- `Objects` in the `Part::FeaturePython` `BOPTools.JoinFeatures.FeatureConnect` and `draftobjects.clone.Clone`
- `OriginFeatures` in `App::Origin`
- `Shapes` in `Part::MultiFuse`
- `Views` in `TechDraw::DrawPage`

There are others not listed.

Now, assuming I want an expression that selects the first "child" in a container, I must use an absurd chain of conditionals to be flexible. Here is a partial example from my current project, where `<<ClipStripConfig>>.Configuration_Preset` is an `App::PropertyXLink` that the user sets to point to a container (ideally of any type they prefer) that is expected to have a first child be a spreadsheet.

`<<ClipStripConfig>>.Configuration_Preset.TypeId == <<App::DocumentObjectGroup>> ? <<ClipStripConfig>>.Configuration_Preset.Group[0] : <<ClipStripConfig>>.Configuration_Preset.TypeId == <<App::Part>> ? <<ClipStripConfig>>.Configuration_Preset.Group[0] : <<ClipStripConfig>>.Configuration_Preset.TypeId == <<App::LinkGroup>> ? <<ClipStripConfig>>.Configuration_Preset.ElementList[0] : False`

There is of course `OutList`, but it also contains objects referenced by property expressions, so is of no use. What we need is a universal name that can be used to list ordered children of any container. Perhaps call it **`Children`**?

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-05-19:
> Perhaps call it **`Children`**?

Great offer. I would also like to add that the “parent”-“children” connection under ideal conditions should work in both directions. Perhaps call it **`Parent`**?
  But right now it only works from parent to child. (Otherwise the problem of circular dependency arises)
Perhaps this is a “system” limitation and cannot be implemented. But if we manage to do this, it would be a revolution!

maxwxyz, 2024-05-19:
Related: https://github.com/FreeCAD/FreeCAD/issues/12796

Roy-043, 2024-05-19:
Are you aware of the `OutList` property?